motioneye-project/motioneyeos #1737: Difference partition between 20181129 and 20181130

tigernero79, 2018-12-08:
The dev version 20181130 already has the change to the new partition? 

because I just mounted pre-release image of 20181129 in which you indicated to rewrite image, now I wonder between the release 20181129 and the release 20181130 what is the difference?

both have the new partition? 

can I upgrade from 29 to 30?

ccrisan, 2018-12-15:
@tigernero79 yes, every release starting with 20181129 has the new partition layout and the upgrade should work seamlessly. If it doesn't, please report back so we can debug the issue and fix it before releasing the stable version.

closed 2018-12-15 by ccrisan

tigernero79, 2018-12-15:
all the updates have been successful, thanks ccrisian, even the last December, I just had to reset the ssh port to any rest that does not maintain it, always reset to 22. but everything resolved by reshaping it.

ccrisan, 2018-12-17:
@tigernero79 good to hear that. Thanks for your feedback.