motioneye-project/motioneyeos #1748: WARNING: mjpg client connection for camera 1 on port 8081 is closed

bob-lee, 2018-12-17:
```
[motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
```
I see above warnings to appear many times in `motioneye.log` and wondering if someone explain what's going on..

I am using:
>  RPI1B
>  MMAL camera
>  motionEye 0.39.3

ccrisan, 2018-12-26:
@bob-lee your RPI1's CPU is probably saturated. Either upgrade to a newer RPi or reduce the resolution and/or framerate.

PhoenixDortmund, 2019-11-29:
@ccrisan 
First of all thank you for this great software!

I have the same problem. I´ve reduced the framerate and the resolution but still no images in the E-Mail Notification. 

I´m using a RPI Zero W
motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911
Attached Pictures Time Span 10 sec.

Thank you!

Trefle10, 2020-09-03:
Found out that this for me occurs when an operation to clean up some old data fails (in my case I had manually deleted these images form the backup NAS so of course they were not there when the motioneye clean up tried to delete them. 
seems like that exception caused the client connection closed error after (until I rebooted it) 
So maybe check if you handle this to not crash the main process on that clean up error. 

Thanks BTW for a great piece of software! 

ERROR: failed to cleanup media files: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_121_216_camera_admin/2020-07-15'
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/cleanup.py", line 86, in _do_cleanup
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 311, in cleanup_media
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 157, in _remove_older_files
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 145, in _list_media_files
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 100, in findfiles
OSError: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_121_216_camera_admin/2020-07-15'
2020-09-01 12:38:17: [motioneye]    ERROR: failed to cleanup media files: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_121_216_camera_admin/2020-07-25'

2020-09-02 01:59:22: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2020-09-03 01:48:27: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081

starbasessd, 2020-09-03:
Out of curiosity which version of motioneye/motioneyeos are you using?
I would like to try to reproduce it

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Trefle10, 2020-09-13:
I do not know anymore.  Updated the entire system since...
It was like 6 months old OS

AntouanK, 2020-10-05:
I have the same log output and the same problem ( never see screenshots attached in emails ).
I tried different intervals but no change.
Any suggestion on how to solve this?

starbasessd, 2020-10-05:
Which version of motionEyeOS?
Which Version of Motion?
Which Version of MotionEye?
(all are available under <General>
Which version of RaspberryPi?
What size microSDCard?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-10-05:
sorry, forgot. 
here you are : 
```
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Debian 9.12
```

running in a docker container ( image `ccrisan/motioneye:master-armhf` ), on hypriot OS
```
$ neofetch 
  `.::///+:/-.        --///+//-:``    pirate@black-pearl 
 `+oooooooooooo:   `+oooooooooooo:    ------------------ 
  /oooo++//ooooo:  ooooo+//+ooooo.    OS: Raspbian GNU/Linux 10 (buster) armv7l 
  `+ooooooo:-:oo-  +o+::/ooooooo:     Host: Raspberry Pi 2 Model B Rev 1.1 
   `:oooooooo+``    `.oooooooo+-      Kernel: 5.4.68-v7+ 
     `:++ooo/.        :+ooo+/.`       Uptime: 20 hours, 58 mins 
        ...`  `.----.` ``..           Packages: 478 (dpkg) 
     .::::-``:::::::::.`-:::-`        Shell: bash 5.0.3 
    -:::-`   .:::::::-`  `-:::-       Terminal: /dev/pts/0 
   `::.  `.--.`  `` `.---.``.::`      CPU: BCM2835 (4) @ 900MHz 
       .::::::::`  -::::::::` `       Memory: 238MiB / 746MiB 
 .::` .:::::::::- `::::::::::``::.
-:::` ::::::::::.  ::::::::::.`:::-                           
::::  -::::::::.   `-::::::::  ::::
-::-   .-:::-.``....``.-::-.   -::-
 .. ``       .::::::::.     `..`..
   -:::-`   -::::::::::`  .:::::`
   :::::::` -::::::::::` :::::::.
   .:::::::  -::::::::. ::::::::
    `-:::::`   ..--.`   ::::::.
      `...`  `...--..`  `...`
            .::::::::::
             `.-::::-`

```

starbasessd, 2020-10-05:
What is your host, and Host OS, and where does Raspbian come into this?
You are showing many layers. Docker is running on what, and what is running
the Raspbian?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-10-05:
I think "hypriot OS" is basically raspbian with docker pre-installed.
I get the images from here --> https://blog.hypriot.com/downloads/

Now the command I use to run motioneye is : 
```
docker run  --name="motioneye"  --hostname="motioneye"     -v /etc/localtime:/etc/localtime:ro     -v /etc/motioneye:/etc/motioneye     -v /var/lib/motioneye:/var/lib/motioneye     --restart="always"     --detach=true   --device=/dev/video0  ccrisan/motioneye:master-armhf
```

starbasessd, 2020-10-05:
Have you tried using RaspberryPiOS, install Docker and use the normal
startup command?
I run docker on several machines (iron and VBox VMs) and have tested
previously on RPi. I am building a RPi4-4GB now on RaspberryPiOS with
latest updates and patches.
I am installing docker [sudo apt install docker.io] from the official
repositories.
I used sudo docker run --name="motioneye"     -p 8765:8765 -p8081:8081
 --hostname="motioneye"     -v /etc/localtime:/etc/localtime:ro     -v
/etc/motioneye:/etc/motioneye     -v /var/lib/motioneye:/var/lib/motioneye
    --restart="always"     --detach=true     ccrisan/motioneye:master-armhf
and it works normally and as expected.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-10-06:
Not sure this would solve the problem. I mean docker and raspbian are not the problem here, right?

Can we run the container with some flag so we can get more verbose logs? Like when it attempts to send an email, to maybe see if something fails?
Also, how is that "Attached Pictures Time Span" work exactly?

starbasessd, 2020-10-06:
When diagnosing issues, you look for commonalities.
In this case, I know the original base systems (RaspberryPiOS and docker.io)
work as expected, and you are reporting issues with unknown (to me)
HypriotOS (with their version of docker) doesn't work as expected.
Since they don't indicate what changes they've made to 'improve' things,
it's possible something(s) they've changed make it not work as expected for
the docker image.
There are several possible solutions:
1) switch to a known working environment
2) switch to their environment, and build your own docker image to work
around the 'improvements'
3) Realize using any locked down system (motionEyeOS or docker image) will
create issues that only certain parameters will be available for
adjustment, vs. a 'normal environment' and not all adjustments may be
available in every environment.
The docker image is not well documented, but does support the standard -v
and -p parameters, but doesn't support adding packages (for example
'cifs-tools') to correct or add features missing from the motionEyeOS or
RaspberryPiOS.image,

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-10-06:
I let it do many motion triggered captions today.
I see that in some of them I do get a screenshot and some I don't.
To me it's obvious that it's the configuration. Either I do something wrong, or I don't understand how it works.
Either way, docker or the OS has nothing to do with it.

I'll ask again because I'm ignorant on the configuration, how does that "Attached Pictures Time Span" work exactly?
Do the motion event have to be above a certain time period?
Do I get a screenshot after X seconds of the capture start?

starbasessd, 2020-10-06:
After motion is detected and ended, it looks for the timespan given and
attaches them in an email alert. However, the more pictures it has to look
throu, in the entire tree, it will start timing out before attaching them
to the emails. It has no effect beyond that. There in another setting in
motioneye to extend the time to look for them.
If you do not wish to try one of my suggestions, I will leave this here for
others to try to help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-10-06:
let's say a motion is triggered, and it captures an event of X seconds.
if the timespan I gave is Y, it looks if X > Y and if so it attaches the screenshot at the Yth second?
I'm a bit confused, sorry if I don't get it. just trying to configure it so I always get a screenshot. ( basically to see and skip false positives and tweak it as I go )

Sangoku, 2021-10-17:
Have this strange bug on reolink 5mp camera. 
Worked earlier. Now i get this:

[00:00:05] INFO: Starting motionEye...
    INFO: hello! this is motionEye server 0.42.1
   DEBUG: using motion version 4.3.2
   DEBUG: using ffmpeg version 4.4
   DEBUG: reading main config from file /data/motioneye/motion.conf...
   DEBUG: loading additional config structure for main, without separators
   DEBUG: additional config section: network
   DEBUG: listing config dir /data/motioneye...
   DEBUG: found camera with id 2
   DEBUG: found camera with id 1
   DEBUG: reading camera config from /data/motioneye/camera-1.conf...
   DEBUG: loading additional config structure for camera, without separators
   DEBUG: reading camera config from /data/motioneye/camera-2.conf...
   DEBUG: listing smb mounts...
   DEBUG: starting motion
   DEBUG: starting motion binary "/usr/bin/motion"
   DEBUG: motion detection disabled by config for camera with id 1
   DEBUG: disabling motion detection for camera with id 1
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: smb mounts started
    INFO: server started
   DEBUG: successfully disabled motion detection for camera with id 1
[00:00:08] INFO: Starting NGinx...
[00:00:08] DEBUG: Requested API resource: http://supervisor/discovery
[00:00:08] DEBUG: Request method: POST
[00:00:08] DEBUG: Request data: {"service":"motioneye","config":{"url":"http://127.0.0.1:28765"}}
[00:00:08] DEBUG: API HTTP Response code: 200
[00:00:08] DEBUG: API Response: {"result": "ok", "data": {"uuid": "383b5596fdd94f3d85d74b8732883b44"}}
[00:00:08] DEBUG: Filtering response using: .uuid
[00:00:09] INFO: Successfully send discovery information to Home Assistant.
   DEBUG: creating mjpg client for camera 2
   DEBUG: mjpg client for camera 2 connected on port 8082
 WARNING: mjpg client connection for camera 2 on port 8082 is closed
   DEBUG: connection closed for mjpg client for camera 2 on port 8082
   DEBUG: mjpg client for camera 2 on port 8082 removed
   DEBUG: creating mjpg client for camera 1
   DEBUG: mjpg client for camera 1 connected on port 8087
   DEBUG: creating mjpg client for camera 2
   DEBUG: mjpg client for camera 2 connected on port 8082
 WARNING: mjpg client connection for camera 2 on port 8082 is closed
   DEBUG: connection closed for mjpg client for camera 2 on port 8082
   DEBUG: mjpg client for camera 2 on port 8082 removed

 DEBUG: getting disk usage for path /share/motioneye/Camera2...
   DEBUG: 200 GET /config/list?_username=admin&_signature=ac25985bf20638d5cb616a0b877e033c1d70decf (127.0.0.1) 18.57ms
   DEBUG: creating mjpg client for camera 2
   DEBUG: mjpg client for camera 2 connected on port 8082
   DEBUG: connection closed for mjpg client for camera 2 on port 8082
   DEBUG: mjpg client for camera 2 on port 8082 removed
   DEBUG: creating mjpg client for camera 2
   DEBUG: mjpg client for camera 2 connected on port 8082
 WARNING: mjpg client connection for camera 2 on port 8082 is closed

closed 2021-10-17 by starbasessd