motioneye-project/motioneyeos #1755: Hub internally shows all feeds, externally only shows 1 feed.

jim2386, 2018-12-27:
### motionEyeOS Version

I am running motionEyeOS version: 0.39.2

### Board Model

I'm  using Rasberry Pi 3 with a USB camera and a pi camera and a Pi zero W with a pi camera. Eventually I'll use the PI3 as a hub only and have multiple pi zero W's with pi cams.

My motionEyeOS unit is connected to the network via wifi
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Orbi Router

I've got a motionpiOS hub on 192.168.1.102 that grabs USB and Pi cam from the Pi3 (hub board) and I have a Pi zero W on 192.168.1.101 that has a pi cam.

When I am inside my home network and go to 192.168.1.102,  I can see all three cameras.

I've got port forwarding working on the hub Pi. However, when I go to my external IP, it only shows one of my cameras and not all three. Is there a way to see my 3 cameras all on one page outside my home network on one page, or do I need to forward each camera on their own individual ports (8081,8082, and 8083 for example).

I'd really like to see my 3 camera hub view outside my home network if possible.

Thanks for the awesome software!

McTucker, 2018-12-29:
Hallo Jim,
Is the problem solved?
What changes did you made?

McTucker, 2018-12-31:
Hallo Jim,

If there's nothing to add, you can close this treath by yourself, Just click on the button "Close".

[image]

ccrisan, 2019-01-10:
@jim2386 you probably added your cameras as Simple MJPEG cameras or you are trying to access them directly (not via motionEye).

Either you forward ports 808x to each of them from your router, or you add them as Network Cameras, in which case, the motionEye port is enough.

closed 2019-01-10 by ccrisan