motioneye-project/motioneyeos #1757: Shutdown not possible while python script is running

superfino, 2018-12-29:
Hi,
I've a Pi Zero running current MotioneyeOS. With a small pyton script, i trigger every second a Pir-sensor to switch maybe a light on. This script is started at startup in userinit.sh. Unfortunatelly with this script running, the shutdown button doesn't work anymore - reboot works well. Killing the script manually shutdown will work again. Any idea what's going wrong here? Thanks in advance for any hint.
Jo

### motionEyeOS Version
I am running motionEyeOS version: motionEyeOS 20180627

### Board Model
I am using the following board/model: Raspberry PI Zero W

### Camera
My camera model is: original Raspberry Pi camera

### Network Connection
My motionEyeOS unit is connected to the network via: wifi

### Peripherals
I am using the following peripherals that I consider relevant to this issue: none

### My Python script:
#!/usr/bin/python
import RPi.GPIO as GPIO
import time
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BOARD)
GPIO.setup(18, GPIO.IN) #Read output from PIR motion sensor
GPIO.setup(16, GPIO.OUT) #Relais on/off
while True:
i=GPIO.input(18)
if i==0: #When output from motion sensor is LOW
#print "No intruders"
GPIO.output(16, 0) #Turn Relais off
time.sleep(1)
elif i==1: #When output from motion sensor is HIGH
#print "Intruder detected"
import os #Run timecheck.sh Script
os.system("sh /data/etc/timecheck.sh")
time.sleep (30)

### Log Files
I consider the following log files relevant to this issue: motioneye.log

2018-12-29 13:39:12: [motioneye]     INFO: hello! this is motionEye server 0.39.2
2018-12-29 13:39:19: [motioneye]     INFO: cleanup started
2018-12-29 13:39:19: [motioneye]     INFO: wsswitch started
2018-12-29 13:39:19: [motioneye]     INFO: tasks started
2018-12-29 13:39:19: [motioneye]     INFO: mjpg client garbage collector started
2018-12-29 13:39:19: [motioneye]     INFO: smb mounts started
2018-12-29 13:39:19: [motioneye]     INFO: server started
2018-12-29 13:40:09: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087
2018-12-29 13:40:19: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087
2018-12-29 13:40:40: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087
2018-12-29 13:40:45: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2018-12-29 13:40:45: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2018-12-29 13:41:06: [motioneye]     INFO: shutting down
2018-12-29 13:41:06: [motioneye]     INFO: executing "/sbin/poweroff"
2018-12-29 13:42:06: [motioneye]    ERROR: mjpg client for camera 1 on port 8087 error: [Errno 104] Connection reset by peer
2018-12-29 13:42:36: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087
2018-12-29 13:42:46: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087
2018-12-29 13:42:56: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087
2018-12-29 13:43:36: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8087

superfino, 2019-01-02:
Hi Ccrissan, indeed, it's not a question of time. The system doesn't shutdown at all although it shows the shutdown screen. After an refresh it's still there.

ccrisan, 2019-01-11:
@superfino is this something that is 100% reproducible?