motioneye-project/motioneyeos #1758: Upload Media Files doesn't close FTP connection after upload image

McTucker, 2018-12-29:
Hallo Calin,

First, my compliments for this fine programm, I installed it yesterday and I'm impressed.
Also my thanks for doing all this work for us ...

After uploading the image, the ftp connection stays open, because it is not ending with a QUIT.
Maybe you can implement this missing QUIT?
[image]

29/12/2018 18:05:06.828 [3248] Client connected from 37.47.34.103.
29/12/2018 18:05:06.828 [3248] 220 Welcome to the BasDataServer on TILANUSS
29/12/2018 18:05:06.906 [3248] USER xxxxxx
29/12/2018 18:05:06.906 [3248] 331 Password required for gutowo
29/12/2018 18:05:06.969 [3248] PASS *********
29/12/2018 18:05:06.969 [3248] 230 User successfully logged in.
29/12/2018 18:05:07.031 [3248] CWD /
29/12/2018 18:05:07.031 [3248] 250 "/" is current directory.
29/12/2018 18:05:07.109 [3248] TYPE A
29/12/2018 18:05:07.109 [3248] 200 Type set to ASCII
29/12/2018 18:05:07.172 [3248] PASV
29/12/2018 18:05:07.172 [3248] 227 Entering Passive Mode (192,168,0,27,4,0)
29/12/2018 18:05:07.313 [3248] NLST
29/12/2018 18:05:07.313 [3248] 150 Opening ASCII mode data connection for directory list.
29/12/2018 18:05:07.313 [3248] 226 Transfer complete
29/12/2018 18:05:07.422 [3248] CWD log
29/12/2018 18:05:07.422 [3248] 250 "/log" is current directory.
29/12/2018 18:05:07.500 [3248] TYPE A
29/12/2018 18:05:07.500 [3248] 200 Type set to ASCII
29/12/2018 18:05:07.563 [3248] PASV
29/12/2018 18:05:07.563 [3248] 227 Entering Passive Mode (192,168,0,27,4,0)
29/12/2018 18:05:07.719 [3248] NLST
29/12/2018 18:05:07.719 [3248] 150 Opening ASCII mode data connection for directory list.
29/12/2018 18:05:07.719 [3248] 226 Transfer complete
29/12/2018 18:05:07.828 [3248] CWD motioneye
29/12/2018 18:05:07.828 [3248] 250 "/log/motioneye" is current directory.
29/12/2018 18:05:07.906 [3248] CWD /log/motioneye
29/12/2018 18:05:07.906 [3248] 250 "/log/motioneye" is current directory.
29/12/2018 18:05:07.984 [3248] TYPE I
29/12/2018 18:05:07.984 [3248] 200 Type set to BINARY
29/12/2018 18:05:08.063 [3248] PASV
29/12/2018 18:05:08.063 [3248] 227 Entering Passive Mode (192,168,0,27,4,0)
29/12/2018 18:05:08.219 [3248] STOR 1705.jpg
29/12/2018 18:05:08.219 [3248] 150 Opening BINARY mode data connection for file transfer.
29/12/2018 18:05:08.406 [3248] 226 Transfer complete
 
Normally there will be a QUIT so the user gets logged out.

damianromanowski, 2019-02-04:
Are you positive that the connection remains open? To me it appears that wget just terminates the connection without sending a QUIT command beforehand.

There's no point in sending extra data to ask the server to terminate your session when you can do it yourself. This _might_ violate the original [FTP RFC](https://www.ietf.org/rfc/rfc959.txt), but as long as your data is intact, who cares?

damianromanowski, 2019-02-04:
The FTP session is handled by [GNU Wget](https://www.gnu.org/software/wget/) and not by MotionEye, I suggest you go complain there instead.

McTucker, 2019-02-04:
I'm not getting files, i'm putting (UPLOADING) files.
Take a good look at the print screen and read the subject please.

ccrisan, 2019-02-04:
@McTucker I'll see what I can do.