motioneye-project/motioneyeos #1760: Fast network camera with ZeroCam

stoffemi, 2019-01-01:
Im trying to use some devices i have with the ZeroCam with IR sensors as fast network camera, but i read that FNC only works with standard CSI camera.

Why is it limited to only the standard Pi CSI camera?
What would it require to get it working with the ZeroCam with IR for example.

### motionEyeOS Version

I am running motionEyeOS version: 20181209

### Board Model

I am using the following board/model: Raspberry PI Zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: ZeroCam with IR sensors

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

* None

### Log Files

I consider the following log files relevant to this issue:

 * None

jasaw, 2019-01-01:
ZeroCam connects to the CSI port, so it's a standard CSI camera, should work in FNC mode.

jasaw, 2019-10-30:
Closing issue due to lack of follow up.

closed 2019-10-30 by jasaw