motioneye-project/motioneyeos #1761: Freeze if view saved pic or video on motion alert.

giuse320, 2019-01-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

giuse320, 2019-01-02:
Hi, I'm using motionEyeOS 20180627 and it normally does its job well.One big problem I found is that when I look at the last photo or video saved while there is a motion alert the system freezes.If I refresh the page from the browser (there are several attempts) I return to watch live streaming but motion no longer saves photos or videos without warning.Unfortunately this always happens to me when there is more need to monitor the area ...What can I improve in the configuration?Sorry for spelling, I use google translator.

jasaw, 2019-01-03:
Do you have lot of media files saved on the device? How long did you wait before declaring it frozen?

What kind of hardware are you running it on? How is it connected to your network? Wifi or Ethernet?

Have you tried logging in via SSH and monitor CPU load? What does `top` command show you when it freezes?

davisstu, 2019-01-03:
This is an issue which a number of people are suffering from - it looks like there is something broken in motion in builds after October which gives you the issue that you are seeing.

It has been reported [here](https://github.com/ccrisan/motioneyeos/issues/1730) and [here](https://github.com/ccrisan/motioneyeos/issues/1729). Awaiting acknowledgement of the issue and a possible fix.