motioneye-project/motioneyeos #1762: lindenis - the perfect board for MotionEye(os) ?

MrJBSwe, 2019-01-02:
http://www.lindeni.org/lindenis-v5.html
https://www.cnx-software.com/2018/08/31/lindenis-v5-allwinner-v5-sbc-4k-smart-linux-camera/

I guess it would take some effort to fully run MotionEyeOs, but it seems to have lots of interesting features...and you can always add an ai-stick...

As always fast boot and low powermodes is not so obvious how to accomplish...but still

McTucker, 2019-01-07:
Maximum image of 720p
Standard no wifi
Much, much bigger then a Pi Zero ;-)
Too expencive $74 US

MrJBSwe, 2019-01-09:
>Maximum image of 720p
support for x2 4k sensors ....Sensor: Sony IMX317, 1/2.5”, 8MP

**The hole point to buy it is the embedded vision engine.**
EVE: is a embedded vision engine.
CVE: is a compute vision engine.

& hw support for
VPU: H.264/H.265 4K@30fps Encoding and Decoding
IPU: Dual ISP 13M@30fps + 8M@30fps, Fisheye, Stitching

>Too expencive $74 US
It's very cheap once you try do this with an raspberry you will probably need an FPGA to do it

ccrisan, 2019-01-11:
@MrJBSwe I'm pretty sure those high-end video specs are actually reached when using hardware acceleration, which is probably not going to be available for ffmpeg/Motion.

MrJBSwe, 2019-01-11:
>ffmpeg/Motion

gstreamer support which should allow for lots of custom plugins
https://github.com/lindenis-org/ldcam-gst

But why not let the AIE do the heavy lifting and for a few select pictures add custom stuff ?
( H265 & 4K seems otherwise to be beyond reach for a small systems and on top of that face/object detection )

**Goodies I like**

AIE (AI Engine) Architecture – Built-in with intelligent analytics acceleration engine with support for motion detection, perimeter defense, video diagnosis, face detection, flow statistics. Supports binocular depth map."

360 Panoramic Stitching with Hardware Acceleration
Lens distortion correction and fisheye correction; Binocular stitching (360° panorama，180° wide angle"

    EVE – Embedded Vision Engine
        Designed for rigid target (such as human face, head, shoulders and etc.) detection
        360p detection speed is more that 30fps
        Minimum single-image detection of 64×64 pixel
        Maximum 4K resolution input and internal scaling, supporting interesting region detection

    CVE – Computer Vision Engine
        Designed for moving target (car, pedestrian, animal and etc.) detection
        D1 resolution is up to 10fps
        Minimum 8×8 pixel for small object detection
        Maximum image of 720p
        Supports various intelligent analysis applications
        Perimeter intruder detection supports maximum warning region numbers of 32
        Perimeter intruder detection can classify people and car by combining calibration algorithm
        Warning line detection supports maximum trip wires of 32-line
        Warning line detection supports single/dual direction detection

code 
https://github.com/lindenis-org/mpp_sample

Anyway I have ordered one and will evaluate it....

MrJBSwe, 2019-03-07:
EVE in the new beagle bone AI seems to be same as EVE in lindeni v5 !?
https://beagleboard.org/ai