motioneye-project/motioneyeos #1768: Need help with crontab

McTucker, 2019-01-07:
Made a crontab with: crontab -e
It didn't exist yet, but now it contains:
`
# Minute   Hour   Day of Month       Month          Day of Week        Command
# (0-59)    (0-23)     (1-31)    (1-12 or Jan-Dec)  (0-6 or Sun-Sat)
  /1       *          *         *                  *            echo "`date`" >> /data/log/cron_test.log
`
The log isn't get updated.
Somebody knows what I'm doing wrong?

motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS 20181209

McTucker, 2019-01-07:
It had to be
*/1 * * * * echo "date" >> /data/log/cron_test.log

closed 2019-01-07 by McTucker

ashishbindal93, 2019-08-16:
Hi
I need a little help related to this topic.

I want to add command in crontab to run every 15 mins to update time from NTP server on my local network. I am unable to find any command for date updation.

Someone please help

jasaw, 2019-08-16:
MotionEyeOS already supports NTP. You can enable NTP from the web interface and point it to your local NTP server. That is the proper way of maintaining time on your device rather than jumping the time every 15 minutes.

ashishbindal93, 2019-08-16:
Hi thanks for such a quick reply.
I am already following that way of doing it.

My problem is that I have several cameras built with raspberry motioneyeos all pointing to my local NTP server running on another raspberry on same network.

My system is cut-off from internet so I have attached a RTC to the raspberry running NTP server.
All is working fine, but sometimes the cameras start too early than NTP server and then cameras show wrong time "01-01-1970".

Cameras don't correct time after boot.

All IPs on the network are fixed.
 
I want my cameras to poll time every 15 mins from my local NTP server. Crontab seems a good option. Please help.

Chiny91, 2019-08-16:
It may be common but it is horrible to twiddle time every 15 minutes.   How about running _ntpdate_, just once, as a command when your motioneyeos has booted, to get away from 1970, then back to ntp, if your ntp server really does work without a net connection.

ashishbindal93, 2019-08-18:
Hi 
I have tried your suggestion but dont know why it is not working.
Please see the output of the commands and guide me.
[image]

Chiny91, 2019-08-18:
It looks as though you have ntp running on motionEyeOS, so ntpdate objects.   Not tested; I suggest you try another port (the -u option) or stop ntp first, use ntpdate, then re-start ntp.   But really, ntp should correct a large time difference on startup (ISTR that motionEyeOS uses ntpd -g), so just stopping/starting ntpd should do it.   Just booting motionEyeOS on to a running ntp server should work.   I'd concentrate on that problem; ensure your R Pi ntp server is running first, ideally always running, as it takes a small number of hours for ntpd to settle.

jasaw, 2019-08-18:
@ashishbindal93 When you get year 1970 date, can you run `ps | grep ntp` command to see if ntpd is running?
ntpd should be running in the background, and with `-g` option too. There's a bug in ntpd where it crashes silently: https://github.com/ccrisan/motioneyeos/issues/1468 You could be running into this bug too.

Chiny91, 2019-08-19:
@jasaw How about replacing the buggy ntp with [ntpsec](https://www.ntpsec.org) ?    Smaller footprint, active development so bugs likely to get fixed, tidier than a watchdog for ntp failures.   I've had ntpsec running since 2016, with 2 or 3 motionEyeOS instances reliably sync'ing against it.   I see ntpsec is now packaged in Debian Buster, so that probably says something for stability.   Acknowledged it may simply be too much work to replace ntp in motionEyeOS.

jasaw, 2019-08-20:
I have no experience with ntpsec, but from the documentation, it looks like a good replacement for ntpd. @ccrisan Should we evaluate ntpsec?

ashishbindal93, 2019-08-22:
@Chiny91 your solution worked.

Steps I followed:
1. crontab -e
2. */5 * * * * ntpdate -u 80 192.168.1.200

Thank you so much everyone for help.

Need little more help :)

Whats the command to shutdown the camera ( motioneyeos raspberry pi ) ?
How can I instruct the cameras from master device( another raspberry pi connected to the same network ) to do the same ?