motioneye-project/motioneyeos #1769: Network Share with FreeNAS SAMBA

jangxbanger, 2019-01-08:
I am running motionEyeOS version: motionEyeOS 20180627
I am using the following board/model: Pi Zero W 1.1.
I am using the following type of camera: Pi Zero W Camera Module
My motionEyeOS unit is connected to the network via:Wifi

Hello, I'm attempting to add a network share to my MotioneyeOS. 
The Share I have setup is a Samba share from me FreeNAS server. I have verified it will mount in other unix distros as well as a windows machine. I have allowed guest access and have attempted with my credentials to mount the drive but it just will not mount. 

I have attached a few screenshots please let me know if you need anything else from me.
 
[image]

[image]

RZFX, 2019-01-09:
The output of "mount -a" tells you that there is no directory such as /home/jangadmin/mnt/camera.
As you verified yourself with "ls /home" you can clearly see that there is really no directory.

Please create the directory first before mounting it.

`mkdir /home/jangadmin/mnt/Camera`

Let me know if it worked out for you!

RZFX, 2019-01-09:
Also consider creating the ".smbcredentials" File in your home directory instead of writing your plain text credentials in the fstab file.

`vi /YourHomeDirectory/.smbcredentials`

`username=<username>`
`password=<password>`

Save and set correct permission

`chmod 600 /YourHomeDirectory/.smbcredentials`

In fstab

`//<server>/<share> <mountpoint> smbfs credentials=</path/to/.smbcredentials> 0 0`

ccrisan, 2019-01-11:
@jangxbanger see https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares

jangxbanger, 2019-01-24:
@RZFX @ccrisan I'm unable to make any directories due to it being a read only file system

RZFX, 2019-01-24:
> @RZFX @ccrisan I'm unable to make any directories due to it being a read only file system

Please have a look at the FAQ below:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file