motioneye-project/motioneyeos #1771: Repeating login prompt on iOS

RZFX, 2019-01-08:
Hello community,

I'm facing the issue on my fresh install that I get a repeated login prompt on my iOS Device(iPhone 7).
It doesn't matter what internet browser I use but when I enter my correct login information the window disappears and almost immediately appears again.

[image]

It is important to mention, that I access the web-interface over a public ip-address. 

On my laptop and android device everything works fine.
Also opening the local ip-address on my iPhone displays me the login screen.

[image]

I already tried clearing my browser cache and so on...

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: Raspberry Pi Zero W).

### Camera

I am using the following type of camera: MMAL.

My camera model is: Raspberry Pi Camera Module V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Log Files

I don't know which log file would be responsible for this issue.

jangxbanger, 2019-01-08:
I personally would try a different mobile browser. I had a similar issue with a publically accessible IP of mine in safari.

RZFX, 2019-01-09:
Thanks for your answer!

I tried chrome but it acts the same as safari...

What browser did you use?

RZFX, 2019-01-11:
However it just started working over a subdomain pointed towards the dyndns of my router...

closed 2019-01-11 by RZFX