motioneye-project/motioneyeos #1772: Raspberry Pi Camera goes offline

rullywowr, 2019-01-09:
Cannot get MotioneyeOS camera to be stable on RaspberryPi Zero W. Camera goes offline very consistently in a matter of hours. Can login to the UI but the camera just is shown as a Grey box with a "no" sign crossing it out. Only a reboot seems to bring the camera back.

Tried:
- Ensuring the power supply is correct and ample (tried several)
- Storing the files locally (and also on the network)
- Reimaging the card with fresh installs of MotionEyeOS
- Device has a static IP set in MotionEyeOS

Seems to do this on multiple versions of identical hardware, as I have several (RPi Zero W and Rpi NoIR Camera)

### Preliminary Docs

Confirmed

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS 20181209

### Board Model

Raspberry Pi Zero W

### Camera

Raspberry Pi NoIR Camera

### Network Connection

WiFi (2.4 GHz)

### Peripherals

None

### Log Files

I consider the following log files relevant to this issue:

boot-2.log
dmesg-2.log
messages-2.log
motion-2.log
motioneye-2.log

 
[boot-2.log](https://github.com/ccrisan/motioneyeos/files/2741290/boot-2.log)
[dmesg-2.log](https://github.com/ccrisan/motioneyeos/files/2741291/dmesg-2.log)
[messages-2.log](https://github.com/ccrisan/motioneyeos/files/2741292/messages-2.log)
[motion-2.log](https://github.com/ccrisan/motioneyeos/files/2741293/motion-2.log)
[motioneye-2.log](https://github.com/ccrisan/motioneyeos/files/2741294/motioneye-2.log)

McTucker, 2019-01-09:
I also had poblems in the beginning.
But sinds I changed the connected V2 Noir as an MMAL Camera, vc.ril.camera, it is stable and the temperature is a little bit lower.
I also restart the Pi Zero at 04:02 AM.
I don't use mvies and motion detection, because it's to much work for the Pi Zero.
Only pictures are uploaded every 5 minutes with SFTP (FTP is not working correctly...)

jasaw, 2019-01-09:
How many cameras do you have connected to your RPi Zero W? You have 2 cameras configured on your system, and we know the 1st camera is the Pi Cam (MMAL interface), and the 2nd camera is via v4l2 interface. Is this 2nd camera a USB cam?

This log shows a 2nd camera has been added, which is a v4l2 interface.
>2019-01-03 11:59:29: [motioneye]     INFO: adding new v4l2 camera with id 2...
2019-01-03 12:00:05: [motioneye]     INFO: removing camera config file /data/etc/thread-2.conf...

This log shows your system has been configured for 2 cameras, but failed to open the MMAL cam.
>Jan  3 11:59:32 CatFudCam user.notice motion: [-1226321016:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Jan  3 11:59:32 CatFudCam user.notice motion: [-1226321016:motion] [NTC] [ALL] config_camera: Processing camera config file thread-2.conf
Jan  3 11:59:32 CatFudCam user.err motion: [1:ml1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
Jan  3 11:59:32 CatFudCam user.err motion: [1:ml1] [ERR] [VID] vid_start: MMAL cam failed to open
Jan  3 11:59:32 CatFudCam user.warn motion: [1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 

If you have a USB cam attached, please remove it and test again. Remove its configuration too.

If you only have a Pi Cam, it means you have a configuration problem. If that is the case, can you please provide step by step instructions on how you end up with bad configuration so we can identify whether this is motionEye bug or user error.

rullywowr, 2019-01-09:
I only have the single  NoIR Cam connected. I did not configure a second camera. I downloaded the image and burned it to the card using Etcher. Added wpa_supplicant.conf and that was it. 
Not sure why there would be two cameras? Never had any other cam hooked up.

jasaw, 2019-01-09:
I suggest you flash the image again, add your wpa_supplicant.conf but leave the configuration as is, don't change anything. You should be able to see the camera feed by default. Let it run for a day, see whether you get the same problem.

zakuma666, 2019-01-13:
I have the exact same error and my cam is also noir via mmal vc.ril.camera.

Should I open A New issue or is it okay to hang on here with my logs?

jasaw, 2019-01-13:
@zakuma666 Please attach your logs here. Please provide more info on your config like resolution, frame rate, saving movies to local file, was OMX encoder used. Is it an intermittent issue or is it triggered by movie recording? How often do you see the problem? Which motionEyeOS version? Did you try an older version, like release from June?

rullywowr, 2019-01-14:
Update: As of 9:33AM EST (24h later) the image is still broadcasting OK on my setup. I have changed no settings except added the wpa_supplicant.conf file. Is there something you suggest I should try?

jasaw, 2019-01-14:
@rullywowr Alright, now you configure the camera the way you want and re-test, but please document everything that you have done, step by step so we can reproduce the problem.

rullywowr, 2019-01-15:
Checked camera this morning at 12:15 PM EST and the camera unfortunately is not available and shows the "no sign". UI is still accessible. 

Log of MotionEyeOS Changes I did yesterday after the 24 hour (burn in period). I made no other changes.

10:08 - 1/14/19 - Logged in using admin and default blank password
10:11 - 1/14/19 - Changed admin password, created Surveillance Name/Password and changed camera name. Then clicked Apply
10:13 - 1/14/19 - Enabled expert settings and clicked apply
10:15 - 1/14/19 - Changed Resolution to 1024x768 and clicked Apply
10:17 - 1/14/19 - Enabled Movies and changed only the maximum time to 300 seconds - clicked Apply

Here are the files after the camera is not detected. Left the state alone and changed nothing else. Let me know if I can provide any more info.

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2760688/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2760689/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2760690/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2760691/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2760684/boot.log)

jasaw, 2019-01-15:
>2019-01-14
15:11:21: [motioneye]  WARNING: 403 GET /login/?_=1547478675507&_username=admin&_signature=...
15:16:51 meye-dd618a00 user.notice motion: [-1226239096:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
15:50:44 meye-dd618a00 user.err motion: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
15:51:37: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
15:51:44 meye-dd618a00 user.err motion: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!
15:51:50: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED

Looking at the logs, I presume you are 5 hours behind UTC, so your config changes were applied at around 3pm UTC. I've merged the relevant events from multiple log files to give an overview of the series of events.

Here's my understanding of the events.
- 3:11pm UTC : User logged in.
- 3:16pm UTC : Applied new configuration, so motion software was restarted.
- 3:50pm UTC : motion thread-1 died.
- 3:51pm UTC : motion couldn't restart thread-1. MotionEye couldn't get the video stream anymore.

This look very much like a motion software bug to me. Can you please attach your `thread-1.conf` file here? While you're at it, can you please reboot your device, enable debugging from the web interface, and attach your log files again after the camera stops working.

I suspect this is the same bug as reported here: https://github.com/ccrisan/motioneyeos/issues/1730

rullywowr, 2019-01-16:
Thank you. Yes, I am UTC -5 for my particular timezone (Eastern Standard Time).

Attaching thread-1.conf to this file (renamed with .log extension for upload purposes)

I have enabled debugging, rebooted motioneyeos, and will reattach log files again when the camera stops working.

[thread-1.conf.log](https://github.com/ccrisan/motioneyeos/files/2764280/thread-1.conf.log)

rullywowr, 2019-01-17:
Camera stopped working again. Here are the latest log files. Let me know what else I can provide please.

[boot.log](https://github.com/ccrisan/motioneyeos/files/2766855/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2766856/dmesg.log)

[motion.log](https://github.com/ccrisan/motioneyeos/files/2766857/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2766858/motioneye.log)

jasaw, 2019-01-17:
@rullywowr I can confirm that your issue is exactly the same as #1730 and @ccrisan is already building a nightly dev image to workaround the issue. We also have a potential fix that I'm currently testing.

rullywowr, 2019-01-20:
Problem is now fixed! Camera is stable. Thanks for all your hard work.

closed 2019-01-20 by rullywowr