motioneye-project/motioneyeos #1789: Cannot add network camera using rtsp

mainport, 2019-01-21:
Hi there,

I am having the following problem: When i want to add a network camera using the rtsp stream i'm unable to add the device. I'm sure that i am using the correct URL and passwords (See information below).  When i try to add the camera with the correct rtsp stream but with a blank or empty password i can add the device.

After looking in the debug files the problem looks to be with the rtsp identifier
[image]
[image]
[motion (1).log](https://github.com/ccrisan/motioneyeos/files/2780670/motion.1.log)
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/2780671/motioneye.1.log)

er that is not being send by foscam

I  have attached screenshots of url that i use and are tested with VLC.

### motionEyeOS Version
motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS 20190119

### Board Model
Raspberry PI 3B+

### Camera
Network camera foscam fi9853ep

### Network Connection
My motionEyeOS unit is connected to the network via: WiFi and Ethernet

### Peripherals

### Log Files
I have attached the logs and i think that the problem is that after usccesfull connection it doesn;t get a rtsp netcam identifier

2019-01-21 23:16:11: [motioneye]    DEBUG: listing cameras
2019-01-21 23:16:11: [motioneye]    DEBUG: testing rtsp netcam at rtsp://192.168.103.5:88/videoMain
2019-01-21 23:16:11: [motioneye]    DEBUG: connected to rtsp netcam
2019-01-21 23:16:12: [motioneye]    DEBUG: no rtsp netcam identifier
2019-01-21 23:16:12: [motioneye]    DEBUG: 200 GET /config/list/?_=1548108970003&scheme=rtsp&host=192.168.103.5&port=88&path=%2FvideoMain&username=&password=&proto=netcam&_username=admin&_signature=b017d93b754ebfd85b86be7330055528c0bed2bc (192.168.103.57) 1021.36ms
2019-01-21 23:16:15: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.27ms
2019-01-21 23:16:16: [motioneye]    DEBUG: listing cameras
2019-01-21 23:16:16: [motioneye]    DEBUG: testing rtsp netcam at rtsp://192.168.103.5:88/videoMain
2019-01-21 23:16:16: [motioneye]    DEBUG: connected to rtsp netcam
2019-01-21 23:16:17: [motioneye]    DEBUG: no rtsp netcam identifier
2019-01-21 23:16:17: [motioneye]    DEBUG: 200 GET /config/list/?_=1548108974903&scheme=rtsp&host=192.168.103.5&port=88&path=%2FvideoMain&username=camera&password=&proto=netcam&_username=admin&_signature=6d765a7da8d0153132a3c1d83cfe2cfb13ac8e3a (192.168.103.57) 1018.36ms
2019-01-21 23:16:20: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.26ms
2019-01-21 23:16:25: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.36ms
2019-01-21 23:16:31: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.33ms
2019-01-21 23:16:36: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.18ms
2019-01-21 23:16:41: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.33ms
2019-01-21 23:16:46: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.32ms
2019-01-21 23:16:51: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.16ms
2019-01-21 23:16:56: [motioneye]    DEBUG: serving log file "motion.log" from "/var/log/motion.log"
2019-01-21 23:16:56: [motioneye]    DEBUG: 200 GET /log/motion/?_username=admin&_signature=28e51b8efd60f7dfa5d3c233ad8fca35b2d43fcb (192.168.103.57) 6.77ms
2019-01-21 23:16:56: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.26ms
2019-01-21 23:16:57: [motioneye]    DEBUG: serving log file "motioneye.log" from "/var/log/motioneye.log"

ctuxboy, 2020-11-16:
I have the same problem.
Can login, to the foscam in my LAN also adding the IP camera to MotionEyeOS
_rtsp://192.168.0.211:88/videoMain_ and adding login credentials without problem
But can't see a video stream
Using Raspberry Pi 3 B+
Version: motionEyeOS 20200606

starbasessd, 2020-11-16:
I had to downgrade, or upgrade to dev20201026.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctuxboy, 2020-11-16:
Hi thx for the info. I will try this 😀

niko0d, 2021-01-23:
I have same problem