motioneye-project/motioneyeos #1791: Issue with using special characters in camera passwords

eckolaw, 2019-01-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (0.39.3).
Motion version 4.0
Debian 9.7

### Board Model

I am using the following board/model: (ProxMox VM).

### Camera

I am using the following type of camera: (IP Network Camera).

My camera model is: (trendnet IP321PI).

### Network Connection

My motionEyeOS unit is connected to the network via: (POE ethernet Hardline).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (mnt smb shares on windows server via fstab for storage)

### Log Files

I consider the following log files relevant to this issue:

motioeye logs giving the following error:

vid_v4lx_start: Failed to open video device /dev/video0: No such file or directory

this caused grey screens and to lockout cameras.

The found fix by troublshooting was the special characters are not supported in the new update.

eckolaw, 2019-01-26:
> Where exactly is your special character placed? If it's a network camera,
> then the message /dev/video0 should be irrelevant. Could you please help us
> reproduce this issue, maybe without revealing your real credentials?

Thank you for the quick response.
I removed @ and $ from passwords.
The password was working perfect before 12 35 am 1/24/2019
Thank you,
J

eckolaw, 2019-01-26:
The characters were at the beginning of the passphrase

ccrisan, 2019-01-26:
@eckolaw the passphrase being what? The password to the network camera? The WiFi password? motionEyeOS admin password?

eckolaw, 2019-01-26:
The network camera password. It being passed from motioeyeos stopped
working. Works fine on camera