motioneye-project/motioneyeos #1794: MotioEyeOS - RPI zero - camera - Synology - Surveillance Station 8.2

dimgod59, 2019-01-27:
Hello, 

i can't add the RPI'cam in Surveillance station.
But i can stream with MotionEyeOS and JEEDOM but not Synology
My camera is not ONVIF ( generic Amazon 5mp IR in CSI )
Please help me for the instructables

Dim

dimgod59, 2019-01-30:
I answer myself having found the solution: it can help others :)

in the box "type" choose "streaming http" and enter the ip of the camera in the format: 
**IpCam:8081/test**

the /test is important otherwise it does not work.

source: https://www.andysblog.de/synology-surveillance-station-nicht-unterstuezte-kamera-anbinden

LangerFeen, 2019-05-12:
> I answer myself having found the solution: it can help others :)
>
> in the box "type" choose "streaming http" and enter the ip of the camera in the format:
> **IpCam:8081/test**
>
> the /test is important [...]

Thank you,
This was very helpful ;)

lowrisk75, 2019-07-05:
Don't manage to add mine 
http://myuser:@myPassword@10.0.1.7:8081
This work fine in safari but not with surveillance.

Also not sure to understand the video patch (/test) what is that suppose to point to?

dimgod59, 2019-12-19:
try with streaming identifiant (not admin) : 
streamingUser:streamingPwd@ipCam:8081/test  
 I don't know why "/test" but it work !