motioneye-project/motioneyeos #1796: Date and time jump back to the epoch

m00k12, 2019-01-29:
Version 20190119.

Rpi Zero W (two different devices), with MMAL Cameras.

One device is configured to NTP, the other HTTP.  Oddly the time and date jumps back to 1970 on both, seen in the saved video filenames and video overlay text.

This wasn't seen in prior releases, so looks like it could be a regression.

m00k12, 2019-01-30:
I just noticed I've been using a static IP address config - it's not that doesn't run DHCP and then somehow doesn't run the time/date check or something silly is it?

ccrisan, 2019-01-30:
@m00k12 having static IP shouldn't affect your system negatively. However judging from your logs, are you sure your system wasn't rebooted those many times, each time starting from 1970 (which is what it does) and adjusting the time as soon as it can?

m00k12, 2019-01-30:
@ccrisan I rebooted it a bunch of times to see if it would pick up the correct time and date.  It didn't.

So I've switched back to DHCP, and now it *is* getting the time and date on both devices.

And so back to static IP, and I realise that I'm an idiot!  The default gateway in the UI was wrong (I'm not sure where it gets the defaults from) - hence no ability to get time/date over the network.

Sorry for the noise on this - it looks like user error.

ccrisan, 2019-01-30:
@m00k12 I'm glad you sorted it out!

closed 2019-01-30 by ccrisan