motioneye-project/motioneyeos #1798: USB Extention Cable Doesn't Seem To Work With Camera

Montegro, 2019-01-30:
I bought these [USB Extention Cables](https://www.amazon.com/gp/product/B07DVVRT8B/ref=ppx_yo_dt_b_asin_title_o01__o00_s00?ie=UTF8&psc=1) from Amazon thinking I could use them to get a bit more distance between my Raspberry Pi3B and the webcams I'm using, that way I could get them at some better different angles. That said, when I attempted to test one of them out with my single camera, I wound up finding out that for some reason my Pi can't recognize the Camera when set up with them. I was wondering if this might be something I'm just ill equipped to handle on my own and have been wracking my brain against for a little while now. Attached are my logs. I found a point where it runs into an error saying it couldn't tell if maybe the cable is bad in the dmesg.log file, but I'm not sure if I'm maybe just confused or not in that case. It's worth noting the cables work, I've used them on my desktop since, and I plugged the camera I was testing back into the Pi after it failing to recognize the camera attached to it, so there might be a success message in there somewhere. Any help would be appreciated.
[messages.log](https://github.com/ccrisan/motioneyeos/files/2814693/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2814694/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2814695/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2814696/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2814697/motioneye.log)

jasaw, 2019-01-30:
Your dmesg log shows this:
>[Wed Dec 31 19:00:08 1969] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[Wed Dec 31 19:00:09 1969] usb 1-1.2: new high-speed USB device number 9 using dwc_otg
[Wed Dec 31 19:00:09 1969] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[Wed Dec 31 19:00:09 1969] usb 1-1-port2: attempt power cycle
[Wed Dec 31 19:00:10 1969] usb 1-1.2: new high-speed USB device number 10 using dwc_otg
[Wed Dec 31 19:00:12 1969] usb 1-1.2: unable to read config index 0 descriptor/start: -71
[Wed Dec 31 19:00:12 1969] usb 1-1.2: can't read configurations, error -71
[Wed Dec 31 19:00:13 1969] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[Wed Dec 31 19:00:13 1969] usb 1-1-port2: unable to enumerate USB device

Looks like a bad USB cable to me, or the pi is not capable of driving such a long USB cable.

Montegro, 2019-01-30:
Would you happen to know of any USB Extension Cables that are known to work with the Pi and the OS?

Montegro, 2019-01-30:
Attempted it again with the cable, kept getting variations of this group of messages:
`[Wed Dec 31 19:30:07 1969] usb 1-1-port2: attempt power cycle
[Wed Dec 31 19:30:08 1969] usb 1-1.2: new high-speed USB device number 57 using dwc_otg
[Wed Dec 31 19:30:09 1969] usb 1-1.2: device not accepting address 57, error -71
[Wed Dec 31 19:30:10 1969] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[Wed Dec 31 19:30:10 1969] usb 1-1-port2: unable to enumerate USB device`
Each iteration kept growing the longer I left it in there, where it kept changing the USB device number. Seems it can't identify the extension cable, shuts down the port, powers it back on and tries again. These cables are new, and work on other machines so they're not bad cables...

jasaw, 2019-01-31:
There are a lot of factors that may affect a long USB cable, things like PCB traces, common-mode chokes, overall noise on the PCB, etc. A long cable may be working perfectly, but doesn't guarantee that another device will work with it.
Short cables are less susceptible.

Montegro, 2019-01-31:
Thanks for the explanation, as these were 9 foot cables, would something like 6 foot cables work? Or would I have to go even shorter than that?
If you don't know, that's fine, I'm just genuinely curious, and a more defined answer might help others later.

jasaw, 2019-01-31:
I tried 3 meters cables on my pi and didn't have any problem with it. Rather than attaching a USB cable on your 9 foot cable, try attaching something else like USB pen drive just as a test.

Montegro, 2019-01-31:
That worked

> [Wed Dec 31 20:19:28 1969] usb-storage 1-1.3:1.0: USB Mass Storage device detected

> [Wed Dec 31 20:19:28 1969] scsi host0: usb-storage 1-1.3:1.0
> [Wed Dec 31 20:19:29 1969] scsi 0:0:0:0: Direct-Access     Generic  Flash Disk       8.07 PQ: 0 ANSI: 4
> [Wed Dec 31 20:19:29 1969] sd 0:0:0:0: [sda] 16081920 512-byte logical blocks: (8.23 GB/7.67 GiB)
> [Wed Dec 31 20:19:29 1969] sd 0:0:0:0: [sda] Write Protect is off
> [Wed Dec 31 20:19:29 1969] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
> [Wed Dec 31 20:19:29 1969] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
> [Wed Dec 31 20:19:29 1969]  sda: sda1
> [Wed Dec 31 20:19:29 1969] sd 0:0:0:0: [sda] Attached SCSI removable disk
> [Wed Dec 31 20:19:35 1969] usb 1-1.3: USB disconnect, device number 60

So the 9 foot cables work with a thumbdrive, but they fail to identify a webcam?

jasaw, 2019-01-31:
I wouldn't be surprised if it's the webcam that doesn't support long USB cable. Possible causes are bad PCB layout on the webcam, or lack of noise filtering (chokes), or unterminated/badly-terminated USB data lines on the webcam. This is getting into analog electronics, and I'm no expert in this subject.
What brand is the webcam?

Montegro, 2019-01-31:
Logitech, the C920

Montegro, 2019-02-01:
From what I've looked up online, people have have success with the C920's working with 6 Foot Cables on their Desktops, apparently with decreased performance of 9 Foot Cables and above. Supposedly for most cameras a combined cable length of above 16 Feet can cause serious issues with the cameras themselves. Just sharing a bit of what I learned.

closed 2019-02-01 by Montegro