motioneye-project/motioneyeos #1801: Feature Request:  Add mysql client to motioneyeos image

discarn8, 2019-02-02:
I currently write server stats to a mysql db (CPU / video temperatures, cpu load, mem, etc.)

I'd like to be able to do the same from my RPis running motioneyeos.  Would it be a reasonable ask to add some sort of sql client to the motioneyeos image, allowing writes to a sql db?

Alternately, I understand that motion already allows for sql writes upon detection / events.  Has anyone utilized that built-in sql feature to write to a sql db for other items, such as for my request?

ccrisan, 2019-02-03:
The answer is no, because:

 * adding such a function to motionEyeOS would increase the image considerably
 * very few users would actually find this useful
 * there are other means to achieve similar functionality,  through webhooks

closed 2019-02-03 by ccrisan

discarn8, 2019-02-03:
Copy that. Thank you for the reply, sir (and the great work, you do)