motioneye-project/motioneyeos #1803: Morion EyeOs update 02032019 crashes system

hooch55555, 2019-02-03:
I just recently updated my camera os to the 02/03/2019 version. It hangs on the update screen until the software crashes. I ended up dumping the previous version back onto the sd card. Just wondering what the issue might be?

Bryan

jasaw, 2019-02-03:
@hooch55555 Which version of motionEyeOS did you upgrade from? Did you read the notes on the release page:
>note: do not attempt to update to this version automatically! Due to the different partition layout, you'll need to either rewrite the OS image from scratch (recommended) or use the following procedure to manually upgrade your system.

DARKFiB3R, 2019-02-05:
Oops, I've made the same mistake. :(

hooch55555, 2019-02-05:
I wish there was a quick fix

RichardWallis, 2019-02-06:
Caught me out too!
For future is there a way that the autoupdate process can check if updating to a braking change new version will fail, and stop you doing it.

It's all great that the release notes warn you  - unfortunately you are not looking at the release page when you click Software Update Check and then yes

ccrisan, 2019-02-06:
@RichardWallis I'm really sorry about this.Future versions have the possibility of stopping an upgrade if incompatibilities are detected.

jasaw, 2019-02-06:
@ccrisan I think a lot more users are going to get caught out the same way as @RichardWallis . Maybe we could tag the release a bit differently such that old versions won't pick it up as a new version, then we filter out the old tags in the [UpdateHandler function](https://github.com/ccrisan/motioneye/blob/28a663e8fa774e5c2f9aa4337affdf3fa879a9b9/motioneye/handlers.py#L1882) so newer releases continue to work.

hooch55555, 2019-02-06:
Pleaseeeeee. I’d rather be able to autoupdate instead of manually
installing. I’m leaving mine on a 2018 version because it works.

Bryan

discarn8, 2019-02-07:
I, also, failed to read the release notes and just happened to check for updates and updated all 5 of my cameras at once - hence, I'll be re-installing 5 cameras ;-)

Lesson learned, I suppose.

ccrisan, 2019-02-07:
I think a better way of solving this mess is to release an intermediate version that uses the old partition layout and that never updates to anything automatically, ever again. Changing tags and versioning format has a relatively large impact at this moment.

@jasaw what do you think?

jasaw, 2019-02-07:
@ccrisan I see two scenarios here where users might get caught out:
1. Users who upgrade regularly. They upgrade as soon as a new release is out. Your proposed idea of releasing an intermediate version will work for them.
2. Users who upgrade once a year. When they do upgrade, we might have a few releases out by then, and they would have missed the intermediate release.

ccrisan, 2019-02-08:
To be honest, I (we) spent a serious amount of time trying to find a solution for this issue. There's no silver bullet for this problem and we have to take a decision.

Users who have difficult/limited access to their boards will not upgrade without checking for release notes or minimal research on possible problems.

Users who have easy access to their boards will have to intervene manually, one way or another. And the more I think about it, the more it makes more sense to flash the OS image from scratch than running a series of painful commands that won't guarantee a successful upgrade.

This being said, the decision is to:

* keep versioning the way it is
* move tedious upgrade steps to a separate wiki page
* refer future releases to this wiki page

closed 2019-02-08 by ccrisan

silversword411, 2019-05-13:
...wish I'd seen this before upgrading all 5 cameras. IMHO if an "update" is going to cause bootloops it should EOL the current firmware version and start a new chain. Any 1-click button Update should never reliably brick a device.