motioneye-project/motioneyeos #1806: DEV20190204 upgrade fails

tigernero79, 2019-02-05:
I tried to update my release dev20190131 to 20190204 as I usually do via terminal but I get this error, is there any way to solve it?

```
[root@Pi_Zero_W ~]# fwupdate upgrade https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20190204.img.xz
upgrading to https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20190204.img.xz
downloading...
downloaded custom
extracting...
cat: can't open '/data/.fwupdate/root_info': No such file or directory
mount: mounting /dev/loop4 on /tmp/fwupdate_root failed: Invalid argument
```

tigernero79, 2019-02-05:
root@Pi_Zero_W sbin]# curl https://raw.githubusercontent.com/ccrisan/motioneyeos/dev/board/common/overlay/sbin/fwupdate -O /sbin/fwupdate
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--   100 15540  100 15540    0     0  39744      0 --:--:-- --:--:-- --:--:-- 39846
curl: (3) <url> malformed

tigernero79, 2019-02-05:
Tester ok on
# cd /sbin
# curl https://raw.githubusercontent.com/ccrisan/motioneyeos/dev/board/common/overlay/sbin/fwupdate -O

But error Now is

ccrisan, 2019-02-05:
@tigernero79 my mistake, `curl` needs `-o` (small oh).

You might need to release some resources left over from previous tries (mount points, loop devices):

    umount /dev/loop*
    losetup -d /dev/loop3
    losetup -d /dev/loop4

Then repeat the upgrade command.