motioneye-project/motioneyeos #1808: Motioneye and Google Drive (again)

Bonesteel, 2019-02-05:
Sorry for the repeat of prior posts, but I reviewed all of those and did not find a resolution. I had the Google Drive upload working on a prior release, so I believe I have followed the proper process before in the past, but it is not working now. I do get the accessing the upload succeed message, but no files are written. Files are written to my local RPi sd card. I have attached the log file and a screen shot. Thanks 
[motioneye(5).log](https://github.com/ccrisan/motioneyeos/files/2833876/motioneye.5.log)
[image]

Bonesteel, 2019-02-05:
I owe you an apology, and feel pretty stupid. I'll explain in hopes that it will help others who happen to do the same thing. 

Motioneye, and the Google Drive upload function are working perfectly fine. I was looking to see an update of the appropriate folder(s) in the Finder on my Mac OS X MacBook Pro. At some point the sync and backup activity/app became disabled on my MBP and those updates were not displayed. I looked at the existing files and nothing had changed. I in correctly assumed no update was happening. I have re-established those connections and all is as before.

My apologies for wasting your time.

JB

closed 2019-02-06 by ccrisan

keithmartinkinyua, 2020-03-27:
I am also having a problem with motion eye uploading images t googledrive, i have followed alll the steps as per the guide.
What could be the possible problem ?