motioneye-project/motioneyeos #1813: motioneyeos releases

freddy64, 2019-02-07:
Hello,
what does "dev" stand for in dev20190204 latest Pre-release ? and what does " nightly-dev" mean ? does it mean the camera will work better at night in low light conditions.
Thank you

popcicle2003, 2019-02-07:
No. "Dev" stands for developer build. It is a pre-release of a stable build. Nightly-dev means that it is a rough build of the previous build that the developer has roughly changed "overnight" (not really though). Not will work better at night. If you want the newest version go with the nightly, but if you want a more stable build go with the newest (non-beta, pre-release, nightly-dev) release.

ccrisan, 2019-02-07:
Asked and answered :)

closed 2019-02-07 by ccrisan