motioneye-project/motioneyeos #1815: motioneyeos lastest release, no contrast, brightness and sturation settings

freddy64, 2019-02-07:
Hello,
I have just installed the latest release 20190119 and pre-release dev20190204 on my raspberry pi(s) zero w using module camras v2 8MP. The brightness, contrast and saturation settings are no longer available from the video device drop down menu. Is this normal? since these settings were available with motionyeos 20180627 that i was using before. The only available setting is the Automatic Brightness, which I don't think I can use with my cameras.

A word of thanks for the motioneyeos softwear as i couldn't have setup security cameras in my house without it.

evga, 2019-02-08:
I have the same problem with my usb camera (Logitech C270) on Raspberry pi 3
I'm using the latest motionEyeOS image

> motionEye Version 0.39.3
> Motion Version 4.1.1
> OS Version | motionEyeOS 20190119

I can manually change the brightness, contrast etc... with SSH using the v4l2-ctl command
but not from the web interface, the only thing available is Automatic Brightness

Attached is the output of v4l2-ctl --all
Let me know if I can do something else to help you fix it.
[v4l2-ctl.log](https://github.com/ccrisan/motioneyeos/files/2846255/v4l2-ctl.log)

ccrisan, 2019-02-08:
Guys, you are right. Recent versions of `v4l2-ctl` display a hex number when listing controls. https://github.com/ccrisan/motioneye/commit/546c8181cda7aa61d0273f553f8f46ccd157ec93 should fix this issue in motionEye.

Please check out the next nightly build, if your setup permits using nightly builds.

closed 2019-02-08 by ccrisan