motioneye-project/motioneyeos #1816: Won't Boot

boa888, 2019-02-08:
I'm insane. I know this because I do the same thing over and over expecting MotionEyeOS to boot. The only thing I get is the rainbow screen and nothing else. I tired the latest image then tried 2 older releases all with the same result. I have used Etcher and Win32DiskImager. I am 100% confident that wpa_supplicant.conf is correct. After every 5 re-images or so times I image Raspbian and it boots fine. 

I have motion up and running including notifications but I thought I would give MotionEyeOS a try but now it is becoming an obsession. I know I am looking right at the answer but I just don't see it. 

Any comments are appreciated

Steve

cameronbunce, 2019-02-09:
Which images are you using and which model Pi?

Decman84, 2019-02-11:
I'm having exactly the same issue the version before the breaking change is fine and works normally. 20180730

boa888, 2019-02-11:
There are too many images out in the wild. I tried using  motioneyeos-raspberrypi3-dev20190209.img.xz but that did not boot BUT this image was found on the 'Supported Device" page 20190119. This one did boot. 

I have been using Motion for some time. I installed it the hard way but it is totally stable, notifications and all. MotionEyeOS for the Zero is supposed to be an 'out of the box' solution but I found it very unstable. It kept dropping wireless and the camera. I will stay with Motion until I see a new release for the Zero.

jasaw, 2019-02-11:
@boa888 Do you have both RPi Zero and RPi3 variants?  Which image did you load on the devices? You tried `motioneyeos-raspberrypi3-dev20190209.img.xz`, but on which device?
I don't think MotionEyeOS does anything special to the wifi compared to Raspbian. Are you saying Raspbian + Motion on RPi Zero has better wifi stability compared to MotionEyeOS on RPi Zero? Can you please elaborate more on your observation and how long did you test it for? We need more information to identify the cause of the issue.

Decman84, 2019-02-12:
Hi,
I tried the pre-release as well as the stable version both failed to boot. I have multiple raspberry pi zero boards. 'motioneyeos-raspberrypi-20190119.img'. I didn't try the raspberry pi 3 version as the zero was only listed against the above version.

Eventually i fell back on an old version that i had on file below (motioneyeos-raspberrypi-dev20180730). Currently have two cameras hooked up. One is working well and the other is dropping the camera constantly... go figure :)
Cheers

boa888, 2019-02-13:
I have not spent much time on the Pi 3B+. I did use an image specifically
for the Pi3 and successfully boot up MotionEyeOS on the 3B+. I attached
Creative WebCam. It was unstable but may have been the fault of the webcam.
For the Zero I finally found the right image (
<https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices> 2010119).
While is stated it was the latest image the same one found here 2010119
<https://github.com/ccrisan/motioneyeos/releases>   would not boot. I also
noted that it was not the latest. Comparing the URLs shows the difference.
One is for the Pi3 and the other for the Pi
 
<https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneye
os-raspberrypi3-20190119.img.xz>
https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeo
s-raspberrypi3-20190119.img.xz
 
<https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneye
os-raspberrypi-20190119.img.xz>
https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeo
s-raspberrypi-20190119.img.xz  This one boots on the Zero
 
However while this image boots it is very unstable. I think 7 minutes was
the record for uptime. I issued commands via SSH but had it on a monitor so
I could monitor error messages. 
 
On the same Zero I have Raspian with Motion installed. It does everything it
is supposed to including email notificatios and it is totally stable 24/7.
MontionEyeOS caught my attention because "could it really be that easy?" but
I could not get it to work.
 
Memory says I tried motioneyeos-raspberrypi3-dev20190209.img.xz and the
release before it with the same result. Both images I tried from here
<https://github.com/ccrisan/motioneyeos/releases>
https://github.com/ccrisan/motioneyeos/releases quit at the rainbow screen.
 
Steve

  _____  

it on GitHub, or mute
6lrYHS_ks5vMfZ4gaJpZM4atchK> the thread.
<https://github.com/notifications/beacon/AVn7k-HsUlKCw5eJOyXmRe4nfusazdljks5
vMfZ4gaJpZM4atchK.gif>

jasaw, 2019-02-13:
>One is for the Pi3 and the other for the Pi
>https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi3-20190119.img.xz
> https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi3-20190119.img.xz
>
>https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi-20190119.img.xz
>https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi-20190119.img.xz This one boots on the Zero

I've checked the URLs that you've posted and they look exactly the same to me. See the above quoted URLs that I've copied from your post, but removed the angle brackets for easy visual comparison. They look exactly the same.

I've checked the links in https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices and https://github.com/ccrisan/motioneyeos/releases for release 20190119, and they are exactly the same too.

Unfortunately I haven't got a spare RPi Zero W to try this out myself. I'll have to buy more RPi Zero W first.

ccrisan, 2019-02-15:
> Unfortunately I haven't got a spare RPi Zero W to try this out myself. I'll have to buy more RPi Zero W first.

I'll run the tests myself and get back to you guys.

@boa888 
> MotionEyeOS for the Zero is supposed to be an 'out of the box' solution but I found it very unstable. It kept dropping wireless and the camera.

We really want to get down to the root cause of all these instability issues that were reported lately. Could you please provide all of your log files from such a setup where things are unstable?

boa888, 2019-02-16:
look closer. the top 2 (no idea how it got doubled) are raspberrypi3
the second set is
raspberrypi  (notice the 3 is missing)

  _____  

https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeo
https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeo

https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeo
<https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneye
https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeo

it on GitHub, or mute
cqyg9wDks5vNJuRgaJpZM4atchK> the thread.
<https://github.com/notifications/beacon/AVn7k4o3PlTDvTVfz_7693xmMODSRd9Jks5
vNJuRgaJpZM4atchK.gif>

ccrisan, 2019-02-17:
I just (re)tested the [20190119](https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi-20190119.img.xz) image on my Raspberry Pi Zero W and it boots just fine. This is the image that's indicated in the [Supported Devices](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices) wiki page as well. 

I am closing this issue since I see no problems booting the latest Zero W image. If you think there is a problem after all with it, please provide detailed information on how to reproduce it.

closed 2019-02-17 by ccrisan

boa888, 2019-02-17:
They are not the same file
[image]

boa888, 2019-02-17:
They are not the same file

  _____  

<https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneye
os-raspberrypi-20190119.img.xz>  image on my Raspberry Pi Zero W and it

it on GitHub, or mute
nTTIGxyks5vOVASgaJpZM4atchK> the thread.
<https://github.com/notifications/beacon/AVn7k20X4QTPhmYw8Q74ESxw88DXp2akks5
vOVASgaJpZM4atchK.gif>

ccrisan, 2019-02-17:
@boa888 what are we talking about here? What are the two URLs that you are comparing?

Decman84, 2019-02-23:
Hey Guys,
Thanks for your help! 
 NFI what changed but this works now :)
Declan

tommy-rk, 2019-12-27:
I'm sorry to reopen this one, but my issue is very similar. I have a Raspberry Pi 4. The current image:
motioneyeos-raspberrypi4-dev20191221.img.xz

I have also tried it with this image:
motioneyeos-raspberrypi4-20190911.img.xz

Both images are not booting for me. When I use etcher with the original image:
2019-09-26-raspbian-buster.zip
my Raspberry Pi 4 boots up nicely.

I am using the same wpa_supplicant.conf for motioneyeos and for raspbian.

motioneyeos just won't boot. The only thing I see is the red light and the yellow light blinking for about 4-7 time slow and one time long.

Let me know, if I should provide you with anything?

cameronbunce, 2019-12-28:
Please mount the boot partition and show us what you have in boot. 

Though I don’t think this is the same issue, it should probably be a new ticket since the pertinent details are different. This is similar, though. 
I’d love to test it myself but I don’t have a pi4

Accepting donations for testing though :)

tommy-rk, 2019-12-28:
I have deleted my last comment.
You can actually reproduce this quite easily without a pi4. Just use a pi3 with a pi4 image :(
Sorry for wasting time.
Using a pi3 with the pi3 image works perfectly fine.

Maybe this should get included in the wiki:
If you pi doesn't boot, you most likely have the wrong image. If you don't have easy physical access to your pi, but can ssh into your machine try this command:

`cat /proc/device-tree/model`

output for me is:
`Raspberry Pi 3 Model B Plus Rev 1.3`

If you are trying the pi4 image with a pi3 then your pi3 won't boot.

Winston70, 2021-04-21:
I've tried  3 or 4 motioneye os files and still can't get ME to boot. I've tried with and without wpa.supplicant.conf. I tried connecting with ethernet and still no dice. Can anyone name with certainty an ME os that works with pi zero? I'm glad I did not pay for this software.

starbasessd, 2021-04-21:
On the back of the PiZero there is some text and a Version #
Can you post that here, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-04-21:
BTW, all versions of motionEyeOS work with PiZeroW (v1.1) 
PiZero (no W) V1.3 and the original PiZero (also no W) only work IF at least one of the following are true
1) You turn off networking with the os.conf file
2) You attach a supported USB WiFi dongle with an OTG adapter to the middle USB port
3) You attach a supported Ethernet Adapter dongle with an OTG Adapter to the middle USB port
The wpa_supplicant.conf file must be created with a Linux friendly utility like NotePad++ in Windows, or nano or vi in a Linux or Mac.
The specification for os.conf is here: https://github.com/ccrisan/thingos/wiki/os.conf
motioneyeOS 20200606 for PiZero is here:
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz

Winston70, 2021-04-21:
Raspberry PI Zero W V1.1