motioneye-project/motioneyeos #1818: Unable to add "Fast Network Camera" to the Hub

Wthebinder, 2019-02-10:
I am running a hub off of a Raspberry Pi 3 Model B+. I have one Zero W running a non-OEM (NOEM) camera and another running an OEM camera as a fast network camera. The NOEM connects to the hub no problem but for the life of me, I cannot get the Fast Network Camera to connect to the hub. Heres what I have tried:

- Local V4L2/MMAL cameras don't work (duh)
- Network Camera says "not a supported network camera"
-Remote motionEye Camera says there are no cameras (despite there being one)
-Simple MJPEG Camera says that it is "not a supported network camera"

Thoughts?

jaud7of8, 2020-06-16:
Wthebinder, did you find a solution for this?
I have the same problem.

starbasessd, 2020-06-16:
Are you able to view the camera stream of the PiZero?
in a web browser on another computer on the same network:
http://<ipaddress>:8081
Can you hook up a monitor to the Zero and see what the screen says?

jaud7of8, 2020-06-16:
Thank you for the reply starbasessd.

I am able to view the stream on another computer on the same network with http://ipaddress:8081

Attached is a picture of the screen of the motioneyesos pi0

This is on a Ubuntu 18.04.4 computer using Firefox
[image]

starbasessd, 2020-06-16:
If you select Type: Network Camera
URL: http://192.168.1.19:8081
Username: admin
Password: 'blank'  << unless you changed it on the Zero
Should pop up the camera info type
and select OK

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jaud7of8, 2020-06-16:
Thanks, that worked.
I had put in the ip address, but did not connect the streaming url with a network streaming camera, that is, I did not add the port number when trying to add the fast camera on the rpi0 with motioneyeos to the motioneyes hub on ubuntu.
Thanks again.

starbasessd, 2020-06-16:
No problem, glad to be of help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2022-01-19:
Closed as completed

closed 2022-01-19 by starbasessd