motioneye-project/motioneyeos #1819: No Image fast Camera dev20190209 pi zero w and pi camera

tigernero79, 2019-02-11:
With the dev20190209 version
if I activate the fast camera mode on pi zero w with picamera 8 mpixel I get out of the ban photo icon when I go to log in because?
[image]

talormanda, 2019-02-14:
At least you got on the newest version! Mine crashed when updating and went into a constant reboot loop.

tigernero79, 2019-02-14:
error to you depends on which version I was going to before updating. it is written that in the new versions have changed the way to partition the system. the problem is the fast camera mode that does not exit the image.

tigernero79, 2019-02-14:
Close for error

ccrisan, 2019-02-17:
@tigernero79 works just fine here in FNC mode on my RPi Zero W. Can you provide more details, such as what exactly does not work? Maybe some logs?

tigernero79, 2019-02-17:
then in normal mode the camera works fine, I talk about the dev20190209 if I change the mode in fnc me out camera symbol not available. I have port 10099 for access and 10100 for streaming. if I call URL for streaming port 10100 I see image in fnc but if I access port 10099 to check it does not exit image but symbol that I have posted.

ccrisan, 2019-02-17:
Could it be that your browser is not happy with the MJPEG stream embedded into the UI? It wouldn't be the first time. The developer console should tell you more.

tigernero79, 2019-02-18:
I also tried from a cell phone but when there is a graphic interface in a fast camera I do not leave the image

 18 10:30:38 Pi_Zero_W user.notice motion: [-1226919032:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Feb 18 10:30:39 Pi_Zero_W user.err motion: [1:ml1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
Feb 18 10:30:39 Pi_Zero_W user.err motion: [1:ml1] [ERR] [VID] vid_start: MMAL cam failed to open
Feb 18 10:30:39 Pi_Zero_W user.warn motion: [1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
Feb 18 10:30:39 Pi_Zero_W user.warn motion: [1:ml1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Feb 18 10:30:39 Pi_Zero_W user.crit motion: [1:ml1] [CRT] [STR] http_bindsock: error binding on any IPv4 address port 10100: Address already in use

tigernero79, 2019-02-18:
2019-02-18 11:21:31: [motioneye]     INFO: mjpg client garbage collector started
2019-02-18 11:21:31: [motioneye]     INFO: smb mounts started
2019-02-18 11:21:31: [motioneye]     INFO: server started
2019-02-18 11:21:32: [motioneye]    ERROR: HTTP 400: Bad Request (unknown operation)
2019-02-18 11:21:32: [motioneye]  WARNING: 400 GET /picture/1/current/?_=1550485291935&width=355&_username=admin&_signature=f8fea77e8c75ba11e42699838a5aa0daad8f9ad8 (37.162.82.147) 50.82ms
2019-02-18 11:21:33: [motioneye]    ERROR: HTTP 400: Bad Request (unknown operation)
2019-02-18 11:21:33: [motioneye]  WARNING: 400 GET /picture/1/current/?_=1550485293081&width=355&_username=admin&_signature=2107cdefef1c03333cc67561c61f58cd0ad7c8c6 (37.162.82.147) 39.36ms
2019-02-18 11:21:39: [motioneye]    ERROR: HTTP 400: Bad Request (unknown operation)
2019-02-18 11:21:39: [motioneye]  WARNING: 400 GET /picture/1/current/?_=1550485294238&width=355&_username=admin&_signature=1e9cd70ae0de467b11d11b972e240d01148492eb (37.162.82.147) 22.28ms
2019-02-18 11:21:40: [motioneye]  WARNING: 403 GET /login/?_=1550485299477 (37.162.82.147

tigernero79, 2019-02-18:
I am going to upgrade previous versions, now I am with dev20190218 and I noticed that streaming url works, but not embed that's why you can not even see it in the gui command, when I launch url http://tigernero.duckdns.org:10099/picture/ 1 / frame /

and I put my credentials always exits a ban with camera, while if I launch the streaming directly

http: // Tiger: xxxxxxxxxx@tigernero.duckdns.org: 10100 /

image is seen

obviously to the possto of the x and my password. but does not recognize embed

reason why you can not even see if you call the gui

tigernero79, 2019-02-18:
practically in fast camera if I recall URL with the management http port 10099 (ex 8080) you can not see image, if I call URL with streaming port 10100 (ex 8881) instead.

shooftie, 2019-02-18:
> At least you got on the newest version! Mine crashed when updating and went into a constant reboot loop.

@talormanda, you too? I thought I was being clumsy but you've made me feel a bit better. I've just had to do a fresh install. I initiated the upgrade on my Zero W via the web UI. It also gave me an excuse to buy a mini HDMI cable and found this:
```
Kernal panic – not syncing: VFS: Unable to mount root fs on unknown-block(179...
```
@tigernero79 @ccrisan If it helps to add some more confusion to the situation, even in the previous release I was unable to see the camera preview on the fast network camera itself when using basic authentication. Browser console:
```
This page includes a password or credit card input in a non-secure context. A warning has been added to the URL bar. For more information, see https://goo.gl/zmWq3m.

[Deprecation] Subresource requests whose URLs contain embedded credentials (e.g. `https://user:pass@host/`) are blocked.
```
`/var/log/motioneye.log`:
```
019-02-18 21:45:56: [motioneye]  WARNING: 403 GET /login/?_=1520526457190&_username=_&_signature=XXXXX (192.168.10.64) 11.10ms
```

tigernero79, 2019-02-19:
@ccrisan the problem in my opinion is that in the fast camera section embed the recall with port intended for httpe not with streaming port reason why no room in the login

the problem is when you call url in fastcamera
 http://cccc.ddns.net:10090/picture/1/frame/

it does not point to streaming port but to http port which has no valid credentials to view the streaming stream itself

ccrisan, 2019-02-19:
@tigernero79 and did it work in previous versions?

tigernero79, 2019-02-19:
@ccrisan I can not remember which version I can not see in the fcn gui anymore. one thing is certain if I set fcn and point to the streaming image port comes out if I use http port like no. But you do not have this problem?

tigernero79, 2019-02-19:
as if you can stream it only with user credentials and not admin.

talormanda, 2019-02-19:
> @talormanda, you too? I thought I was being clumsy but you've made me feel a bit better. I've just had to do a fresh install. I initiated the upgrade on my Zero W via the web UI. It also gave me an [...]

I had 2 pi's that fell victim to this reboot loop. I connected it to HDMI and saw it rebooting every 10 seconds or so. Did not sit well with me at 12AM!

ccrisan, 2019-04-06:
@johny-mnemonic I honestly have no idea how to properly implement/fix this one. On one hand I know credentials listed direct in the URL are not a good idea, but on the other hand I wouldn't know how else to pass them to the server without being prompted for credentials. I'm open for suggestions.

Also please keep in mind that the "Fast Network Camera" mode is not intended to be viewed in motionEye on a regular basis, but rather used by another piece of software, which probably won't display the stream as a MJPEG.

johny-mnemonic, 2019-04-07:
@ccrisan I am not a coder, but what about other authentication methods then login+password? Like some token based ones or some challege/response based ones?
I often see in log that client first tries to connect somewhere and only when it receives 401 (unauthorized) as a reply, it sends the credentials/secret token.
I have been told, this is basically a must, where you are doing a lot of connections, because authenticating for every connection would be too expensive.
So I guess these ways of authentication are pretty common and well documented.

ccrisan, 2019-04-08:
@johny-mnemonic that's what we do in motionEye when *not* in FNC mode. But FNC mode, along with Simple MJPEG Camera mode relies on standard MJPEG over HTTP, in which case we cannot tweak the authentication method. It has to be Basic or Digest, since it has to work with other "clients".

johny-mnemonic, 2019-04-08:
@ccrisan Just to make sure I get it right. You are saying, that the piece of code, that is doing the actual job of streaming MJPEG doesn't support more advanced methods of authentication than Basic (login+password) and Digest?
I just Googled, what Digest authentication is, and it looks like what I was describing. Client goes somewhere and only when it gets 401 from the server, it presents a MD5 hash from login+password to a server.
I would guess, this should work fine. Your web will just need to catch that 401 and use stored credentials to generate and send that MD5 hash, instead of displaying authentication prompt to a user.
But as I am not a coder, I maybe missed some catch on the way ;-)

ccrisan, 2019-04-09:
> You are saying, that the piece of code, that is doing the actual job of streaming MJPEG doesn't support more advanced methods of authentication than Basic (login+password) and Digest?

Yes. Motion and streamEye are the two components that do MJPEG streaming and they only work with standard HTTP authentication.

> I would guess, this should work fine. Your web will just need to catch that 401 and use stored credentials to generate and send that MD5 hash, instead of displaying authentication prompt to a user.

That's what happens. Your browser will prompt for credentials and remember them (until when, that depends on various factors). The problem here is that your streaming credentials have already been entered in the *Normal User* section of motionEye and when you change them, the browser won't update its "remembered" MJPEG credentials as well.

In any case, this particular function of motionEye, where you're viewing the MJPEG stream directly, is not one of the most useful functionalities. It should mainly be used for setup, since it doesn't really do anything useful, like motion detection or recording.

johny-mnemonic, 2019-04-10:
I am sorry, but I don't see what is the issue. 
You can simply notify the user, that when streaming credentials change, camera view won't work, till full page reload. 
I do agree, that viewing the camera in web interface is there mainly for setup, but that means, it is very useful. On the other hand, it would be perfectly fine, to replace the camera view with the button opening camera view in new tab/window where user will have to enter credentials and basically verify the very functionality they want when switching MotionEye to Fast Network Camera.

ccrisan, 2019-05-30:
@johny-mnemonic what we've discussed earlier should be "fixed" by https://github.com/ccrisan/motioneye/commit/5669be2d5b75b56b1fdbc7a675eeea0b72f6882e.

Basically credentials are no longer embedded into the URL. However this will still not work in most cases since browsers usually don't prompt for credentials for resources referred from DOM.

If anyone has any other idea on how to improve this, I'm all ears. Until then, this issue is closed.

closed 2019-05-30 by ccrisan