motioneye-project/motioneyeos #1821: MotioneyesOS use pan tilt

dimgod59, 2019-02-11:
Hi, it’s possible to add a pan tilt systèm in my rpi for a MotionEyeOS ? Just plug and Play ?

cpapplefamily, 2019-09-07:
I might be nice to have these basic scripts included and simply require an enable bit set in a config file. The basic scrips maybe mapped to default GPIO pins

martin1957, 2019-11-15:
Hi, I am using MotioneyeeOs on a Pi-zero which is attached to a pan-tilt head.
This head is controlled by a ESP32 which also provides an AP-server for the Pi-zero to transfer pics  and movies because I use it for animals/birds observation in nature. 
Everything so far is working nicely. 
Now I would like to implement a follow-motion function for the head controller. 
My idea is to have GPIO outputs set from the Pi depending on which pic area the (most) motion is detected (upper third, left third... etc.) to be able to process this in the ESP and move the head correspondingly. I think four GPIO pins for the side areas and one four the center indication would be enough for a nice control.
I saw already that there are some nice functions in motion: 
https://motion-project.github.io/motion_config.html#area_detect
Does anyone have an idea which could help to implement this function ?