motioneye-project/motioneyeos #1822: PiZeroW disconnects after ~1day. no IP connection (SSH, http)

njordan77, 2019-02-12:
Anyone else having issues - i recently started with MotionEye. All works fine on PiZeroW....good connection to an OpenWRT (TP-Link) router with -60dBm signal (rather good).....when trying to reconnect via http after a day or so ..... not reachable anymore...tried SSH...same.

Any idea what i could do...its DHCP connected.

Steven0972, 2019-02-13:
yep same here don't know what going on after, a day the pi's ethernet stop working, tried it with different pi's and al the same.
and the update's don't work for me

jasaw, 2019-02-13:
Can you guys please provide more information about your setup?
- Which motionEyeOS version?
- Both of you observed this on RPi Zero W only?
- Disconnect happens on both Wifi and Ethernet?
- Which router are you using?
- When you say disconnected, have you tried pinging the IP address? Find the IP address from your router.
- What type of camera is attached?

njordan77, 2019-02-14:
Hi, as written above in my case its a Pi Zero W
only WiFi, no external Ethernet Dongle installed.
Router is a TP-Link W940 based on OpenWRT, several other devices (NAS, WebCam from D-Link) worked for years now without any disruption (wireless).
Nothing reacts....PING, SSH, WEB, SMB....as if its not available on IP-Layer.
Even if i cannot believe its related to camera, its a standard NOIR 8MPixel from/for Raspi

ccrisan, 2019-02-15:
@njordan77 can you please attach all of your log files as soon as you can recover them? We need to investigate these stability issues.

njordan77, 2019-02-17:
Hi,
i downloaded all log files from var/log. you find them in the zip file attached. hope log does not clear after a reset, as this log is taken after another unconnect from wifi.....i reset the system on Friday ~6pm......

[test.zip](https://github.com/ccrisan/motioneyeos/files/2872746/test.zip)

ccrisan, 2019-02-17:
@njordan77 I'm seeing a lot of `Setting current date using ntp: failed` messages in your `boot.log`. Can you please try using the HTTP date method and see if the problem persists?

Also, the next nightly build will have wpa_supplicant debugging enabled when `OS_DEBUG` is set. Could you please try it as well (as soon as it will be ready) and turn on the *Expert Settings > Enable Debugging option*? Then, in case of WiFi problems, the `wpa_supplicant.log` should contain more details on the problem.

njordan77, 2019-03-06:
Hi, as a final feedback - i switched to Pi 3A+ and no issues at all anymore. for myself ZeroW was no good and brought a lot of issues - including not able to handle email notifications with attachments.

closed 2019-03-06 by njordan77