motioneye-project/motioneyeos #1824: Use of Dropbox upload indepently of UI file upload.

Pi-DD, 2019-02-14:
Is there a way of accessing the Dropbox upload facility directly from, say, a script or Python program?

I ask because I use it as part of a system monitoring process I have written for my Pi's and would like to run it on the MotionEyeOS system I use.

ccrisan, 2019-02-14:
Most likely not, since it uses internal motionEye stuff. You are probably looking for a dropbox python client, such as the official one: https://www.dropbox.com/developers-v1/core/start/python.

If you need this level of customizations, I recommend using Raspbian + motionEye.

closed 2019-02-14 by ccrisan

Pi-DD, 2019-02-15:
Calvin, thanks for the usual swift response. No problem. As it happens I have other ways of monitoring the Pi that runs MotionEye and I’m happy to live with that rather than try to inject my amateur coding into your nice stable software! 🤨