motioneye-project/motioneyeos #1825: Reboot loop after update from dev20181008 to 20190119

miwallac, 2019-02-15:
I just updated motioneyeos running on an RPi 3 from dev20181008 to 20190119 using the motioneyeos Software Update option in the UI.  It is now stuck in a reboot loop.  The original setup had been running for several months.  Screen cap follows:

[image]

ccrisan, 2019-02-15:
You'll need to rewrite the OS from scratch. As stated, this particular upgrade process is not guaranteed to succeed. We're sorry for the inconvenience.

closed 2019-02-15 by ccrisan

miwallac, 2019-02-15:
Thank you for the quick response.  Thanks to all your hard work, the rebuild process is quick and painless.  Thanks for all you've done!

EpicLPer, 2019-02-21:
tbh this should've been addressed better... I don't always check here before upgrading something and simply use the Update function, and suddenly it won't boot anymore...

jasaw, 2019-02-21:
@EpicLPer See https://github.com/ccrisan/motioneyeos/issues/1803#issuecomment-461940048 for the reasons.

EpicLPer, 2019-02-22:
If @ccrisan already posted a guide on how to (hopefully) update from an older version to a newer one why couldn't he implement that script into the update procedure? If it fails, well it fails, no chance there then but if it succeeds at least there's a tiny chance of people being able to "just update" without having to check the release page first. I don't want to go back here and run multiple commands just to update to a newer release, other projects can do this seamless too so why shouldn't this one be able to do the same?

ccrisan, 2019-02-22:
@EpicLPer because the procedure requires running commands *before* the actual upgrade. And with the previous OS image, we did not have that possibility.

Now, with recent images, we've added support for downloading a new image, extracting it and temporarily peeking inside for pre-upgrade scripts. This is so we don't have situations like this one from now on.