motioneye-project/motioneyeos #1826: Can't turn LED Off HD Pro Webcam C920

mostym, 2019-02-16:
Hi there.  I love your software, though I did I have one question.  How can i turn off the LED on my HD Pro Webcam C920?

Thanks!

ccrisan, 2019-02-16:
No idea. That's something specific to the camera itself, not the V4L2 interface, as far as I know.

closed 2019-02-16 by ccrisan

mostym, 2019-02-17:
Hey @ccrisan when I install just motioneye on raspbian, and then do the following, I can turn off the LED.  Can I please request this as an enhancement?  I would much rather use your motioneyeOS.  Thanks!

```
sudo apt-get install uvcdynctrl
sudo uvcdynctrl -i /usr/share/uvcdynctrl/data/046d/logitech.xml

# LED off
uvcdynctrl -s 'LED1 Mode' 0
```

ccrisan, 2019-02-17:
We have `uvcdynctrl` on motionEyeOS so you can use that.

allthegoodnamesweret4ken, 2019-04-16:
@mostym 
> 
> ```
> sudo apt-get install uvcdynctrl
> sudo uvcdynctrl -i /usr/share/uvcdynctrl/data/046d/logitech.xml
> 
> # LED off
> uvcdynctrl -s 'LED1 Mode' 0
> ```

Did you manage to switch off the LED? If so, care to share how? I have the exact same camera, and am facing the same problem.

alpa2009, 2020-01-23:
knowing this is an old post, but having the same problem: it seems that control option 'LED1 Mode' does not exist, at least it is not listed when listing the available user controls...  was this ever solved? Tnx

sfduff, 2020-04-12:
Using **uvcdynctrl** to turn off Logitech c920 LEDs does work, even with the errors, so long as you don't have the local pi camera interface enabled using Raspi-config.  
As soon as the pi camera interface is enabled, it stops working; disable it and once again!
I have a couple of RPi's using 'both' a pi camera (for night vision) and a Logitech c920 for daytime connected.
I used to be able to operate both from the same pi3b+ using Stretch, but have not been able to do the same with Buster.
I'm guessing it is something to do with address space for the devices, but it is a bit beyond me tbh, and help welcome.

floriansiepe, 2022-06-24:
For those who are facing the same issue. I was able to disable the led by using `v4l2-ctl`.

```bash
# Get the right device
v4l2-ctl --list-devices
# Get the current setting (my device was /dev/video4)
v4l2-ctl --get-ctrl=led1_mode -d /dev/video4
# Disable the LED
v4l2-ctl --set-ctrl=led1_mode=0 -d /dev/video4
```

Paderman, 2023-09-19:
> For those who are facing the same issue. I was able to disable the led by using `v4l2-ctl`.
Thank you for your help 🙂

Where can I find all commands for this camera?
I must e.g. flip the camera but I don't know how does it work.

Is it possible to get and set frame rate and many more?

Thank you in advance.