motioneye-project/motioneyeos #1828: wifi connection never established

talbojak, 2019-02-16:


ccrisan, 2019-02-16:
@talbojak have you checked our [wifi troubleshooting](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems) article.

talbojak, 2019-02-22:
gone through all the points but still wont boot up and connect to wifi

ccrisan, 2019-02-22:
Please try the latest [nightly build](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev), turn *Expert Settings > Enable Debugging* on and attach your `/var/log/wpa_supplicant.log` (from a live system, probably using Ethernet).

talbojak, 2019-06-23:
where would I find this?

jasaw, 2019-06-24:
@talbojak Latest nightly build is here: https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev
Click on the link.