motioneye-project/motioneyeos #1829: Pi Zero W won't connect on reboot.

captaintapir, 2019-02-17:
Having responded to an older thread, I'm now pulling my hair out at trying to get my Pi Zero W working correctly after an upgrade and months of trouble-free motioneyeos-ing.

My issue is that on the second (and subsequent) reboot, it goes into a boot loop, presumably because it cannot connect to the wifi network.  I initially wondered whether it was due to my buggering up the configuration, but it does so even if I reboot as soon as the first boot has completed and I've accessed the web interface.  wpa_supplicant.conf is configured correctly - I can access the Pi on the first boot.  It just breaks on the second boot, even if no configuration changes have been made.

If I then copy wpa_supplicant.conf onto /boot, it'll boot once, and then fail to boot/connect to wifi on the next restart.

I've tried:
- 20192009
- 20190119
- 20181209

Any ideas, please?

ccrisan, 2019-02-17:
@captaintapir 

> These are the logs - they are only visible on the first boot, but don't appear to show up on the micro SD card if I browse it:

What do you mean they are only visible on the first boot? Where are they visible? Do they disappear afterwards?

captaintapir, 2019-02-18:
Thanks for your response.  I can access logs from the web interface, but when I browse to /data/log, the folder is empty after I've turned the Pi off.  Looking at them, it does appear that they are surviving reboots, but for the life of me I can't find them browsing or searching for them.

ccrisan, 2019-02-18:
Look for them on the third, big partition, in the `/log` folder. It's mounted in `/data/log` when the OS is running.

captaintapir, 2019-02-23:
Okay, I've upgraded to the latest dev release, but have had no change to the behaviour.  It boots correctly on the first attempt, and then panics on the second.  Below are two sets of log files - one with a static IP address and one with DHCP.  Other than changing to static in the first set, and the timezone in the latter, nothing has been changed from the default configuration.

Static:

DHCP:

captaintapir, 2019-02-24:
file:///home/pir/Desktop/logs/v2/boot.log
file:///home/pir/Desktop/logs/v2/dmesg.log
file:///home/pir/Desktop/logs/v2/messages
file:///home/pir/Desktop/logs/v2/motion.log
file:///home/pir/Desktop/logs/v2/motioneye.log
file:///home/pir/Desktop/logs/v2/wpa_supplicant.log
file:///home/pir/Desktop/logs/v2/boot.log
file:///home/pir/Desktop/logs/v2/dmesg.log
file:///home/pir/Desktop/logs/v2/messages
file:///home/pir/Desktop/logs/v2/motion.log
file:///home/pir/Desktop/logs/v2/motioneye.log
file:///home/pir/Desktop/logs/v2/wpa_supplicant.log
file:///home/pir/Desktop/logs/v2/boot.log
file:///home/pir/Desktop/logs/v2/dmesg.log
file:///home/pir/Desktop/logs/v2/messages
file:///home/pir/Desktop/logs/v2/motion.log
file:///home/pir/Desktop/logs/v2/motioneye.log
file:///home/pir/Desktop/logs/v2/wpa_supplicant.log
file:///home/pir/Desktop/logs/v2/boot.log
file:///home/pir/Desktop/logs/v2/dmesg.log
file:///home/pir/Desktop/logs/v2/messages
file:///home/pir/Desktop/logs/v2/motion.log
file:///home/pir/Desktop/logs/v2/motioneye.log
file:///home/pir/Desktop/logs/v2/wpa_supplicant.log
file:///home/pir/Desktop/logs/v2/boot.log
file:///home/pir/Desktop/logs/v2/dmesg.log
file:///home/pir/Desktop/logs/v2/messages
file:///home/pir/Desktop/logs/v2/motion.log
file:///home/pir/Desktop/logs/v2/motioneye.log
file:///home/pir/Desktop/logs/v2/wpa_supplicant.log
file:///home/pir/Desktop/logs/v2/boot.log
file:///home/pir/Desktop/logs/v2/dmesg.log
file:///home/pir/Desktop/logs/v2/messages
file:///home/pir/Desktop/logs/v2/motion.log
file:///home/pir/Desktop/logs/v2/motioneye.log
file:///home/pir/Desktop/logs/v2/wpa_supplicant.log

ccrisan, 2019-02-26:
@captaintapir as soon as you switched to static IP, your `boot.log` started complaining:

     * Configuring wireless network: 192.168.1.169/24
     * Setting static gateway to 192.168.1.1: done
     * Setting static DNS server to 192.168.1.216: done
     * Setting current date using http: failed

Are you 100% sure your DNS server `192.168.1.216` does actually work as a DNS server?

captaintapir, 2019-02-26:
@ccrisan, other devices seem to work correctly using it, so I'm reasonably confident.  It's a recursive DNS server using unbound.  Nothing else is causing problems for me.  The other devices have also been configured recently, so they don't have caches of addresses.

Will give it a try using a third party DNS provider and let you know.

ccrisan, 2019-02-26:
@captaintapir could you please run some tests while it is accessible (during the first boot, while with DHCP)? Try to manually set your nameserver in `/etc/resolv.conf` and then run the following command:

    curl -v -s -m 10 -X GET http://google.com 2>&1 | grep Date | sed -e 's/< Date: //'

See if it properly returns the current date.

captaintapir, 2019-02-28:
Thanks.  It does (I'm GMT +11):

[root@meye-a7fa742d ~]# curl -v -s -m 10 -X GET http://google.com 2>&1 | grep Date | sed -e 's/< Date: //'
Thu, 28 Feb 2019 09:43:56 GMT

I did this as soon as it booted, haven't logged into the web interface or done anything else from the shell.

ccrisan, 2019-02-28:
Then could you attach a monitor (and maybe a keyboard) to it and try reproducing the issue? This way we could see what's actually going on when setting static IP and it stops working.

captaintapir, 2019-03-04:
Sorry for the slow response @ccrisan - had some issues tracking down a Mini-HDMI adapter.

I've reflashed the micro SD card and done the following:

> curl -v -s -m 10 -X GET http://google.com 2>&1 | grep Date | sed -e 's/< Date: //'
-> displayed the current time correctly

> reboot

> curl -v -s -m 10 -X GET http://google.com 2>&1 | grep Date | sed -e 's/< Date: //'
-> displayed the current time correctly

Without rebooting, logged into web interface.  Changed the time zone from UTC to Australia/Melbourne, which forced a reboot when I applied the setting.

Upon reboot, hit the problem:
[..]
* starting wpa_supplicant: done (brcmfmac)
* configuring wireless network: dhcp
wifi: disconnected
wifi: disconnected
wifi: disconnected
wifi: disconnected
wifi: disconnected
wifi: disconnected
wifi: disconnected for 10 second, calling panic action
panic: rebooting in 10 seconds (caused by wifi)
wifi: disconnected for 20 second, calling panic action
panic: rebooting in 20 seconds (caused by wifi)

It attempts to ping 192.168.1.1, then spits out a few more 'starting' messages (http date updater crond, sshd, proftpd, smbd, nmbd, motioneye). 

It then continues to boot through the normal sequence of messages, before eventually restarting, and then repeating the process.

The only change is that on the second attempt the 'disconnected for 10 seconds' becomes '40 seconds', the '20 seconds' becomes '50 seconds), the third boot '10 seconds' became '60 seconds', then '20 seconds'  became '70 seconds', but otherwise identical to the first boot.  Then 80, 90, then 100, 110, then 120, 130 seconds.  You get the idea.

On the fourth reboot, it actually reached a login prompt, but then rebooted whilst I was typing some of this message.  On the fifth reboot, I managed to log in until it restarted.  ifconfig revealed that wifi0 wasn't set up correctly, obviously.

I haven't managed to stop the panic-caused reboot, but there is a longer gap between reboots each time, so there's opportunity for some quick fixes - if I knew what to do!

ccrisan, 2019-03-04:
@captaintapir thanks for taking the time to do all this testing. Regarding the incremental rebooting, see https://github.com/ccrisan/thingos/wiki/Panic-Reboots. It's basically a way to prevent wearing out the flash memory card while still rebooting in case of a problem.

If you know your way around Linux, you could try editing `/etc/init.d/panic` and removing the `reboot` command.

But given that you particularly have problems with the WiFi connection, please make sure you have Enable Debugging option turned on in Expert Settings and let's see what your `/var/log/wpa_supplicant.log` has to say.

captaintapir, 2019-03-09:
@ccrisan, thanks for taking the time to help me try to solve this.  Also, thanks for teaching me a little more about init.d

Here's wpa_aupplicant.log (I've removed by SSID from it, but otherwise it's the full thing).
[wpa_supplicant.log](https://github.com/ccrisan/motioneyeos/files/2948093/wpa_supplicant.log)

And the rest of the logs:
[boot.log](https://github.com/ccrisan/motioneyeos/files/2948094/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2948095/dmesg.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2948096/motioneye.log)

ccrisan, 2019-03-10:
Man, this beats me. I assume you added yourself a ping command somewhere in the init scripts:

``` 
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
PING 192.168.1.1 (192.168.1.1): 56 data bytes

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss
```
I can't see why your DHCP would offer you an IP configuration & everything and then won't respond to ping.

What I would do next is to find a USB ethernet adapter and use that to remotely login and try to debug the problem. Alternatively, that could be done using a keyboard and a monitor.

captaintapir, 2019-03-17:
@ccrisan, thanks again. I haven't solved it yet, but I'm looking into it again.   I've downloaded older versions of motioneyeos and the same issue is occurring.  I forget which version I used to use, but as it's for a PiZeroW, I must have tested it this time without luck.

I didn't add the ping command, so no idea where that came from!

I'm now going to try reconfiguring my router to see if this helps, plus using the recent build from a few days ago.  The setup is currently using the router as the gateway, but a RPi with a couple of Pi-holes as the DHCP servers (split by IP address, so there isn't any overlap) and the router's own DHCP turned off.  I'll re-enable it, plus also trying the router's guest network with a simpler password (my wi-fi has the full 63 character limit in use).

captaintapir, 2019-03-17:
Okay, I've had a play with the wi-fi, and it hasn't made much difference.  However, it hasn't made any difference.

Having said this, I have tried a few things and have managed to change the behaviour a little:
- created wpa_supplicant.conf.bak on /boot - it's a backup of 
 - added a line to /etc/init.d/S35wifi:
     cp /boot/wpa_supplicant.conf.bak /boot/wpa_supplicant.conf
- changed to a static IP address, assigned to the same range that Pi-hole has in its DHCP range.

Intention was to recreate wpa_supplicant.conf on boot every time as it works once.  This failed because the filesystem is read-only (no surprise, really).  However, by doing so, it seems to have done something to fix it.  I can now reboot and successfully connect to the network.  I don't understand what has changed enough to fix it, but it seems to be fine now.

Grunds, 2019-03-27:
Mine is in the same boat, Has anyone tried re-flashing to an older build?

captaintapir, 2019-03-30:
@Grunds - I did, but it didn't fix my issue.  In my case, I suspect that it's something to do with the unusual network setup of two self-hosted DHCP servers and recursive DNS servers.  I made a bunch of changes at the same time as I upgraded my motioneyeos installation, which means I can't pin it down to a specific change I made.

I intend to check my backed up configuration file (I think I have one) to see if it lists which version I was using previously and then try again from scratch with that one, as well as just updating the wpa_supplicant.conf with the new details and using my old config for everything else.  Will post again with the result.

Grunds, 2019-04-01:
@captaintapir  I'm running a pihole on my network for DNS. I tried bypassing it, but it seems like the wpa_supplicant is in read only and sudo is not installed so I'm not sure how to change that. 
@ccrisan is there a way to change the read only status of the wpa_supplicant.conf file. I think that is why I can't get my wifi configured as it works over Ethernet.

jasaw, 2019-04-01:
@Grunds `/data/etc/wpa_supplicant.conf` file is **not** read-only. Also, if you want to replace `/data/etc/wpa_supplicant.conf`, you can put your new `wpa_supplicant.conf` file in `/boot` partition and motionEyeOS will do the replacement on the next boot up.

user9023523, 2019-04-08:
@ccrisan I'm getting the same issue with my RPi0w running the latest stable. On the first boot it works fine but on reboots sometimes I won't see the camera on the network. Pulling the plug and plugging it back in usually fixes it. Don't have this issue on my RPi3B+.

ccrisan, 2019-04-08:
I tried to reproduce this problem on my setup (same SSID on 2.4 and 5 GHz), but the Zero W worked just fine, always connecting to the 2.4 GHz. If you guys can play a bit with your WiFI settings (channels, band steering, etc) and see what causes the problem, that would help us a lot.

rwmccoy, 2019-04-13:
Just to complicate the issue, I'm having a similar problem on an Odroid C2 directly attached to wired Ethernet. 
Symptoms are:
Flash C2 Image onto memory card.
Insert and boot C2 first time, menu comes up normally.
Change timezone (no other changes), allow forced reboot. 
C2 no longer responds, can't ping, can't SSH, no web interface, no activity on HDMI monitor.

I can't verify I'm getting the same logs as my Mac appears unable to mount the large partition on the memory card. Unless someone knows a way to do this on a Mac, I'll have to crank up my Debian laptop and read it there. Haven't used it in a while so that may take some time.

user9023523, 2019-04-18:
@ccrisan  I plugged in a display to see what's going on. As soon as I issue a "reboot" over SSH the Zero W starts rebooting but it gets stuck on
`Starting eudev: done` 
so it might not even be trying to connect to WiFi.

ccrisan, 2019-04-18:
Definitely not. The UI actually calls the reboot command.

bachoo786, 2019-04-21:
i am having the same issues here. first boot after installing raspbian stretch lite works fine but if I was to reboot the rpi zero w 2nd or 3rd time it will take good 3 to 4 mins to connect to the wifi 2.4ghz and then make itself available.

user9023523, 2019-04-26:
I fixed this problem by getting a new microSD card and reflashing meyeOS. That particular board now connects reliably. But yes sometimes it does take a while to show up.

ccrisan, 2019-07-15:
Could you please try the latest [nightly dev build](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) and see if the WiFi problem persists?

captaintapir, 2019-07-17:
@ccrisan - thanks, will do.  Won't be until the weekend, probably.

captaintapir, 2019-12-09:
@ccrisan I can't believe it's been nearly five months since I intended testing it "on the weekend" - life got in the way...  But I've done it now.  The good news is that the issue looks to have gone.  I've restarted three times now, and have been able to reconnect each time without a kernel panic.

Thanks for looking into the issue, and apologies for the really tardy response.  I look forward to getting back to using it again!

ccrisan, 2019-12-09:
Thanks for getting back to us. We can close the issue then.

closed 2019-12-09 by ccrisan