motioneye-project/motioneyeos #1830: Won't Boot 2

boa888, 2019-02-17:
I see my last issue (Won’t Boot #1816) has been judged INVALID. I want to make just one more comment.  Now it is probably just me but I would like some input before it is dismissed. 

I downloaded the following MotionEyeOS image from the releases page shown below. I could not get it to boot on the Pi Zero (it did boot on the Pi3 B+)

From the Releases page
https://github.com/ccrisan/motioneyeos/releases

https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi3-20190119.img.xz
****************************************** 
I just happen to trip over the Supported Devices page so I downloaded it and it did boot on the Zero

From the Supported Devices Page
https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices

https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi-20190119.img.xz

I noticed the file URLs were different (pi vs pi3). Once downloaded and compared I noticed the compressed file sizes were different but the decompressed files were the same size. Even though they share these two files have the same release number they are NOT the same file. To be sure I ran the Checksums of the two files. They are different  proving they are not the same. 
[image]

Am I missing something? Either way one would boot and one would not but that could just be me

ccrisan, 2019-02-17:
Why did you download the raspberrypi3 version? The Supported Devices page clearly says the version for the Zero is the raspberrypi one.

ccrisan, 2019-02-17:
Also, why would the two files you just compared be identical? Why do you expect that?

boa888, 2019-02-18:
OK, just mark this as INVALID
 
Accept my apologies. When I did a search "MotionEyeOS Latest Release" I got
to the main page of downloads. I had asked about this an unfortunately I was
told I could use the Pi3.
 
I will cause no more problems
 
thanks
Steve
 

  _____  

it on GitHub, or mute
Z4Cdlieks5vOeFYgaJpZM4a_0zY> the thread.
<https://github.com/notifications/beacon/AVn7k7d3QJSPWJLsQgKpycHfrsxd5ruTks5
vOeFYgaJpZM4a_0zY.gif>

ccrisan, 2019-02-18:
Alright :)

closed 2019-02-18 by ccrisan