motioneye-project/motioneyeos #1839: MotionEye - "invalid credential"

sergioaiello, 2019-02-27:
SOS!
I encountered the "invalid credential"@login issue three times in the last few hours. Every time I had to start from a scratch/formatted SD since I could not figure out the correct username/password combination (I did many many tries, also with their hashes, without success)

My conclusion is that MOTIONEYE inherits the password from the camera (or any other previous login).
In my last try I followed this procedure:

* as I added a new camera (the first of the system and a "network camera"), I was requested to insert camera's username and password, the browser asked to store them and I accepted => immediately motionEye registered that password in the control board.
* i cleaned MotionEye "Admin Password" edit box and clicked on "Apply"
* MotionEye applied and asked me to login. I left the password field empty and it was ok
* I kept on configuring the video device/video streaming/motion detection/movies sections for the new camera
* Click on "Apply", MotionEye applied and asked for credentials: the password field was not empty and if blanked (as it should be) my login was rejected: "invalid credential". Now Login was possible with camera's password .... If I would have logged in to my email box then MotionEye would expect that password? I preferred not testing that when I inserted the email for email notification, but I can do if needed.
  Then I removed the camera and the admin password to test the sequence again.
  Click on "Add Camera" to get the popup window, but in background I saw that MotionEye filled up the password field again. Why??
  After "Apply" the MotionEye asked for login and "password" field was already populated (with 5 dots as it did previously). I blanked and I got the "invalid credential" warning, so again MotionEye wrote not appropriately the password in the database.

Here are the HW/SW info if that any help:
* motionEye Version: 0.39.3
* Motion Version: 4.0
* OS Version: Debian 9.8
* Linux orangepipc2 4.19.20-sunxi64 #5.75 SMP Fri Feb 8 10:29:25 CET 2019 aarch64 GNU/Linux (more detail here: http://ix.io/1BW3)
* Chrome Version 72.0.3626.109 (Official Build) (64 bit)
  Please, do not hesitate to let me know if additional info are needed.
  Thank you for your support!

Jpsy, 2019-03-10:
I am also stuck with "Invalid credentials".
I erased the hash from  `# @admin_username` in `/etc/motioneye/motion.conf`  and restarted the pi. I also deleted browser cookies. But no luck.
It started when I tried to changed the admin password through the web gui. The new password is not working. Nor is the old one. I did not misspell it - always used clipboard. And I cannot seem to reset the password through motion.conf as said above.

Maxduck, 2019-03-18:
Me too, stuck @ "Invalid credentials" .
I was trying to adjust motion settings as it was capturing every 30 secs for 8 hours overnight, regardless of motion being present or not.
Tried emptying browser cache no luck.
There is no log  @ /var/log/motioneye.log, only the motion log file?
running on KDE neon...

tokamac, 2019-05-03:
Same problem here. Hassbian Strech, installed motion 4.2.2 fine, then motioneye with pip2 (to avoid issues with Python3). Worked like a charm until I installed wrong `/dev/video*` for my USB (V4L2 H.264) camera. Was immediately unlogged, and from this moment on I can never log in MotionEye GUI. When I type my ID and password I get "invalid credentials". If I enter again my password, a new blank window opens with the address `http://MYI_P/login/`
I uninstalled and reinstalled motioneye, rebooted each time… motioneye is totally broken. 
Any solution? For example can we erase the file containing that password? Where is it located?

tokamac, 2019-05-03:
@Jpsy retry what you've done but uncomment the `#` before `@admin_username` and `@admin_password`

jsiddall, 2020-07-23:
Confirmed what @okamac wrote.  Even though there was nothing set in my motion.conf for the admin password, and both were commented as below:

# @admin_username
# @admin_password

I could not login on the web UI.  SSH still worked.  So I went into the /data/etc/motion.conf file and uncommented those lines, adding admin after the admin_username:

@admin_username admin
@admin_password

Then I could login again in the web UI and also see the "General Settings" and set a new password.

For some reason, on initial setup, the entire "General Settings" section was missing and after setting some things in "Expert Settings" and hitting apply I could no longer login after reboot.  I am running:

motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911

on a Pi Zero W.

ctrl50, 2021-06-13:
just had this problem and found this issue from google.

Followed what jsiddal said by adding 

@admin_username admin
@admin_password

was able to log back in.

closed 2021-06-20 by starbasessd