motioneye-project/motioneyeos #1840: Slow FPS when using either ngrok or DDNS

ash32152, 2019-02-28:
Hi All,

Just wondering if anyone could advise on the issue I'm seeing, locally on my network MotionEye seems to be working fine, running around 20-25FPS, I have tried to port forward that and access externally using DDNS, ngrok and a VPN to my home network, no matter which option I try, I seem to be stuck with around 3FPS, my dl is around 40MB, my up is around 15MB so I wouldn't have thought they'd have struggled... I'm using a low res (640x480) FPS set at 30 for the cameras themselves, quality is at 50% and I use the RPI solely for motioneye with no motion detection etc, just live preview as I have a NVR that deals with the recording... anyone able to suggest why I get such a low frame rate externally when my upload can more than handle it? Viewing the cameras directly streaming them in HD (1080) its's smooth, just through motion eye I get poor fps externally.

cheers,

Dararii, 2019-12-06:
@ash32152 Can you give a tutorial how to set up ngrok in motioneyeos? Thanks in advance

ash32152, 2019-12-06:
Hi mate,

Yes sure, I will get one wrote up and post it over the weekend! 😊

Cheers,

digiltd, 2019-12-14:
have you tried it without ngrok or ddns, i.e. a regular port forward, just for a short while to test it?

oh and be sure you don't have framerate dimming set to low

>Viewing the cameras directly streaming them in HD (1080) its's smooth

Is that streaming them from outside your home network, if so how?

Dararii, 2019-12-16:
@ash32152 Thank you so much. Looking forward to your tutorial.

lukedukeus, 2021-08-14:
@ash32152  Did you ever find out why?