motioneye-project/motioneyeos #1841: Is it possible to break out of the GUI into terminal?

mhaserodt, 2019-02-28:
I'm running a RPi Zero W headless and as a WiFi access point.  When I want to change the time due to daylight savings, I've been pulling the Pi out, hooking it up to a monitor, and changing it through the terminal.  

Is it possible for me to connect to MotionEye OS like I usually do using WiFi, then get into the terminal that way to save me pulling the Pi out of the project?  Or is it possible to wirelessly SSH into the thing, while it's set as a WiFi access point, and accomplish the same?

ccrisan, 2019-03-01:
I don't see why not. Just access the motionEye UI via its IP address. With the default DHCP settings, it'll be 192.168.27.1.

mhaserodt, 2019-03-01:
I apologize, I'm pretty new to all things Raspberry Pi.

I normally use the IP address (192.168.27.1) which takes me right into the motionEye OS where I can see the realtime video, change settings, etc.   I'm not sure how to get to the command line from there though.

ccrisan, 2019-03-01:
Using SSH or Putty?

mhaserodt, 2019-03-01:
Neither.  I'm just connecting to the Pi like it's a WiFi hotspot then using the IP address in whatever browser.

It sounds like I need to learn more about SSH and Putty and using those with the Pi running as an access point.

ccrisan, 2019-03-02:
Well if you need terminal access, you're probably looking for either SSH or Putty.

closed 2019-03-02 by ccrisan