motioneye-project/motioneyeos #1842: Stuck config change apply? Everythink asks to

cy80rg101, 2019-03-01:
Hi,

Im wanting to check for updates (in particular, I notice the cleanCloud feature was added last month).

However, when I got to "CHECK" for new versions via the admin UI, I get the "Please apply the modified settings first" message, even though I've not changed anything.

I even deliberately made a change, got the red "Apply" button up, applyed that change, then rebooted, but I still get the message on trying to CHECK.

Double frustrating, as I've been trying to write my own "cleanCloud" scripts for ages, without success, so if I can update to that, would be great! :)

I'm running the following release:

motionEye Version	0.39.2
Motion Version	4.1.1
OS Version	motionEyeOS 20180627

cy80rg101, 2019-03-01:
UPDATE: 

Just seen the new upgrade process - ie manual re-image - and have updated to :

motionEye Version | 0.39.3
-- | --
Motion Version | 4.1.1
OS Version | motionEyeOS 20190119

Can edit and change system settings, but trying to add a camera (any - this is a clean install) gives the "apply modifed settings first" message... 

There arent any, and the "Apply" button isnt avilable... 

This is a clean install, on a formatted sd card...

ccrisan, 2019-03-02:
@cy80rg101 can I see a screenshot of your problem?

AIexBV, 2019-03-10:
I have had the same issue. After changing the browser the message was gone. So maybe that was a cache or cookie issue.

sagittarius0815, 2019-03-10:
> I have had the same issue. After changing the browser the message was gone. So maybe that was a cache or cookie issue.

Can confirm. I have had same issue (using chrome). Using another browser (firefox) solved the problem.

ccrisan, 2019-03-11:
Browsers recently decided it is a good idea to trigger change events whenever they autofill usernames and passwords. This issue has been fixed upstream via https://github.com/ccrisan/motioneye/commit/8cb4e46eea834e5737eb61ce160aa735d7377d5a. It will be part of the next release.

closed 2019-03-11 by ccrisan

denisjoshua, 2019-04-15:
I have same problem, but I can't update cause I have the motioneye installed 
via Home Assistant (Hassio) addon. 
Can I solve this browser issue mantaining the same version please ? 
Thanks a lot 
Denis