motioneye-project/motioneyeos #1843: Motion not detected thru Window

ejwjohn, 2019-03-02:
Is there any way to configure Motion to be detected when a camera is viewing and area through glass ie window please?

Thx

JohnW

ccrisan, 2019-03-02:
Perhaps you are looking for the Mask feature?

ejwjohn, 2019-03-02:
Hi,

Thanks for your response, but i tried that and it did not seem to make any difference.

But maybe i am setting the mask incorrectly.

Thx

JohnW

ejwjohn, 2019-03-03:
Hello,

I have continued to investigate this issue and have read that using PIR through windows can be tricky, but using a software solution to measure pixel change works. I have tried  different settings on Motioneye OS and i still cannot get any motion detection outside the glass. The camera i am using is an ELP variety, but that should not make any difference.....

Thanks for you continued support.

JohnW

jasaw, 2019-03-03:
@ejwjohn In most cases, PIR will not work through windows because of the PIR technology, so you are better off with image based (pixel change) motion detection. Even then, certain scenarios will still render the image detection behind window useless, for example high glare as a result of light hitting the glass. My advise is to turn on the "show debug frames" and play with the motion detection threshold to find a good compromise.

ejwjohn, 2019-03-04:
OK,

My set up is ELP camera (V4L2) with Auto Brightness activated and other settings as default for the camera

Frame change is set at 1.6% with Auto Noise detect active Light switch detection at 25% and Despeckle filter active.Mask is on        Editable but i have not selected any areas to ignore. Create Debug is now on and i get the following images, but i am not sure what i am supposed to be looking for.

Thanks

JohnW
[image]
[image]
[image]
[image]