motioneye-project/motioneyeos #1844: "The image xxxxx cannot be displayed because it contains errors"

peppelorum, 2019-03-03:
After a few minutes of using MotionEye the cameras freeze and the live image is replaced by the "no camera" icon. If I then go to the current url of the camera it says: "The image http://<ipnumber and path> cannot be displayed because it contains errors"

But if I visit the camera directly, or use VLC  it works fine.
If I look in motion.log I see nothing special.

Any ideas of what can be the problem?

peppelorum, 2019-03-03:
Odd, after a while MotionEye would show the snapshot and the live feed again. Any ideas why it happened at all?

limones1319, 2020-05-29:
I am having the same issue. Although the camera is recording pictures and saving them to the storage.

Mine doesn't show any live image from the beginning.

blushingpenguin, 2020-08-13:
Same for me (possibly after upgrading ubuntu 18.04 -> 20.04) -- motion continues to work fine (images / videos captured; emails are sent), rtsp stream in vlc is fine, no stream in the motioneye ui

starbasessd, 2020-08-13:
did you run
sudo pip install motioneye --upgrade
sudo systemctl restart motioneye
???

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

blushingpenguin, 2020-08-13:
Yes, and it was running ok (no errors in the logs from motioneye).  There were a few warnings about obsolete configuration values from motion, and after deleting motion.conf it seems to be working fine now, so some sort of configuration issue was the problem.