motioneye-project/motioneyeos #1848: Software restarts crash pi 3B+ & Zero W

marcsgirall, 2019-03-07:
Hi there,

I am on mobile as I write this so I don't have access to the troubleshooting logs, but I can provide them if needed. The issues I'm having are twofold:

First - my Pi Zero Ws & the Pi 3B+ hub I'm using both become unresponsive after 24-48 hours of uptime, sometimes needing multiple power cycles before coming back onto the network & being usable.

Second - the cameras also crash outright when I perform a software reboot (either via the GUI or by SSH, result is the same) & again require multiple power cycles to work.

I'm running the most recent stable build of Motioneyeos & the 3B+ is connected to the router via Cat5e while the Zero Ws are connected over wifi.

Thanks for any feedback you may have!

marcsgirall, 2019-03-15:
Follow-up: the Pi 3B+ has now completely crashed. It no longer starts up when power cycling. Going to have to reflash it, I think.

talormanda, 2019-03-29:
Would check the type of SD card being used here. (manufacturer, class rating, etc)

marcsgirall, 2019-03-29:
Appreciate it - I reflashed & it’s been working flawlessly since. I think there was a weird error in the original flashing. For the record it’s a Class 10 Patriot microsd

-Marc Girard Alleyn

closed 2019-07-15 by ccrisan