motioneye-project/motioneyeos #1849: Home Mode

eldadh, 2019-03-08:
It Would Be great if there could be a "Home Mode" to temporary disable motion detection,
either by API or WebUI.
Each time i disable motion detection or notification all settings are lost.

trees-smoke, 2019-04-30:
Would bluetooth on a phone or checking the ARP tables from motionEye be a good way to toggle this?

moo-im-a-cow, 2019-05-01:
something else that would be a lot more flexible would be a HomeAssistant compatible switch.
(Home Assistant is an open source Home automation software)

it has a wide range of presence detection (including bluetooth, ping, connection to various routers)

Creating a Switch component for that would let Home Assistant control motioneye, which can be linked to the prescence detection, or other events (e.g. even if no motion, start record if the house alarm goes off)

Some other camera manufacturers already have this (e.g. https://www.home-assistant.io/components/amcrest/)

coderpop, 2019-05-11:
ANother option would be ato have motioneyeos able to set or read gpio pins.

If pin(x) goes high then set home mode, if low set normal mode. Similar uses could be turning on/off lights from the pi and having external motion detectors trigger capture mode.

The one really useful thing would be to have on each camera overlay the ability to pause motion detection either fully and manually restore or just for a given time in minutes.