motioneye-project/motioneyeos #1850: Can't acces MotionEyeOS IP when using WiFi

Niklassen, 2019-03-08:
Preliminary Docs
I confirm that I have read the CONTRIBUTING guide before opening this issue. (YES)

I confirm that I have read the FAQ before opening this issue. (YES)

motionEyeOS Version
I am running motionEyeOS version:  motionEyeOS dev20190223 (latest as of now)

motionEye Version | 0.39.4
Motion Version | 4.1.1
OS Version | motionEyeOS dev20190223

Board Model
 Raspberry pi 3 model A+ and tried Raspberry pi 3 model B 

Camera
Microsoft LiveCam Studio

Network Connection
WiFi 2,4 GHZ and tried cable

Peripherals: none

I am using the official raspi power supply, so no undervoltage problems.

I am experiencing a very strange problem. i have used MotionEyeOS on my Model b for a long time, and then upgraded to model A+. it all worked. Then i got tired of the ugly setup, and took it down and waited some days. I have been trying to set it up again for two days now. Now, to the problem. I first flashed the image to my A+ and loaded it with the WiFi PreConfiguration. It's IP showed up on my router settings as usual. But when i try to access the IP it simply leads me to (this site can't be reached) or something. It says the site took too long to respond. Then i did the same thing to my model B and got the exact same result. then i tried to connect my model B to cable, and then it works. even if it uses the same IP. Model A+ doesn't have a LAN input, so i couldn't try it on that. The problem is that i can access it with cable, but not WiFi, even if the MotionEye IP shows up in my router settings. This is strange since the last time i did the exact same thing to both pies it worked great.  I have tried so many different things, like reflashing many times to change small settings. My main goal here is to get it to work on my A+. any thoughts?

ccrisan, 2019-03-08:
Keep in mind that the wired and the wireless connections will have different IP addresses. Also the two boards have different wireless adapters so their IP addresses will also be different.

This being said, while running your tests, expect to have to deal with 3 different IP addresses. If that's not the case, then you have some network configuration issues.

Niklassen, 2019-03-08:
@ccrisan Sorry, I didn’t explain that right. What you are saying is definitely not the problem. If I first use a cable the IP is indeed different than when using WiFi. if I try to turn on wireless connection when accessing MotionEye using cable I will see that the IP changes. But again, accessing it won’t work. But if I then plug it with a cable again, it does work even though it has the same IP. That is what I meant. I admit I wasn’t very clear there. And yes, it uses a different IP when using the other Pi. I was simply using the model B to see if it would work  if I turned on WiFi when using cable instead of preconfiguring it on the A+. That’s probably what caused the confusion. And one other thing to mention is that the signal strength simply isn’t a problem. I can see it is very strong in my router settings when using WiFi. But again, I can’t access it then. Any other thoughts?

ccrisan, 2019-03-08:
Can you access your A+ via SSH/Putty? If yes, see if you find anything interesting through the /var/log files.

Niklassen, 2019-03-08:
@ccrisan Will check after a good night sleep, it’s late here in Norway.

Niklassen, 2019-03-09:
@ccrisan I can access it via SSH and as a surveillance user when using cable.

trongod05, 2019-04-27:
I'm also having this same issue.  If I reboot and try to connect by IP while the pi is on Wifi only, I can't connect through any port or protocol.  I can also not ping the address.  I can see it in my list of clients via the router.  I'm able to actually ping the address from the router but somehow not my desktop. 

Here's the strange part.  If I hook up a display and keyboard, log in locally, then ping the IP of the client I am trying to connect from, I am then able to access the pi from that address only.  Seems like it has something to do with blocking inbound connections unless traffic is initiated toward the connecting client first.  Could this be some sort of iptables issue?

jasaw, 2019-04-29:
@trongod05 Are you running Windows on your desktop? This sounds like routing issue on your desktop. My Windows 10 often has routing issues and I have to manually edit its routing table.

trongod05, 2019-04-29:
@jasaw I’ve tried from different platforms.  Linux, Windows, Mac OS, iOS, etc.  Same problem.  Sometimes it will randomly start working.  Highly doubt it’s a routing issue when all the clients are on the same network.

DaciaLogan, 2019-06-14:
I have similar problem, the wpa_supplicant file disappears after first boot, and the pi will not then re connect to WiFi, but will still work perfectly through ethernet cable.  After the first WIFI boot, the software goes into a loop trying to find a cable connection, and somehow cannot find the wpa_supplicant file, which has become hidden, or possibly deleted.  Looking on youtube, this is a regular problem.  MotionEYEOS becomes unusable on a WIFI connection. .

ccrisan, 2019-06-16:
@DaciaLogan the fact that your `wpa_supplicant.conf` disappears from the boot partition is a good sign: it means that motionEyeOS actually used it. You'll find it in `/data/etc/wpa_supplicant.conf` if you log in remotely to your Pi.

The fact that it struggles with the WiFi connection is a different issue. First off, have you tried Raspbian with the exact wpa_supplicant.conf you provided for motionEyeOS? If not, please do and see if it manages to connect.

DaciaLogan, 2019-06-17:
Hi Calin,
I apologise because I did not make it clear.  The issue for me is that a newly flashed card, with the wpa.supplicant file copied in, works perfectly the first time. It can't connect a second time, and if I then ping the pi, I get ,." Host unreachable. " The same thing tends to happen if I reboot MotionEyeOS.
On your suggestion to try the file with Raspbian, I don't know how to do this, because the pi OS will connect to my WIFI anyway.
I would appreciate your input with this, because it is driving me round the bend!
Many thanks!

David
Mobile 07975 987841

DaciaLogan, 2019-06-18:
Hi Calin,
further news!  wpa_supplicant.conf file does move to the SETTINGS Directory in MotionEyeOS.  It becomes a hidden file and readonly. It also changes to:
ctrl_interface=DIR=/var/run/wpa_supplicant   GROUP=netdev
ap_scan=1
update_config=1
There is no other info, so all network details are deleted,  which means the ssid is now deleted, and the psk.
So the MOTION OS opens the first time, but will not open again.
Is this the problem?
David
Mobile 07975 987841

ccrisan, 2019-06-18:
> further news!  wpa_supplicant.conf file does move to the SETTINGS Directory in MotionEyeOS

What do you mean by "SETTINGS Directory"?

> It becomes a hidden file

What do you mean by hidden? In Unix, files starting with a dot are considered hidden, by convention.

> ...and readonly

That's probably because you're looking on a read-only partition (`/boot`, `/`).

> So the MOTION OS opens the first time, but will not open again.

It's motionEyeOS.

> Mobile 0797*******

I would advise against publishing your personal phone number on public sites, such as this issue tracker. Also I don't do support over the phone.

I would also like to see your `wpa_supplicant.conf` file (replacing sensible data with dummy crap).

DaciaLogan, 2019-06-18:
Hi Calin,
Pls find attached a copy of the exact text used in my wpa.supplicant.conf file.  I have just changed the SSID and PSK details to "xxxx" for this exercise.
I have found this file in the ETC folder, with all the correct details.  There exists however another wpa_supplicant file in the SETTINGS folder, without the SSID and PSK details.
.

David

country=GB
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    ssid="xxxx"
    psk="xxxx"
}

Solderfinger, 2019-09-09:
Hi everybody,
sorry to revive a somewhat older threat but i just stumbled over it, looking for a solution to that exact same problem.
I admit i'm a windoze user (shame on me). The problem is that after writing a completely new image including a working "wpa_supplicant.conf" to the SD card my pi zero w connects to the wifi and works fine. So far so good.
When i try to read this same SD card later on my main windoze machine the "wpa_supplicant.conf" file is no longer there and there is also no hidden file of that name.
I understand from ccrisans posts that after the file has been read it is copied to another directory which obviously is out of range for windoze.

Now my question is:
How do i use this camera in another wifi environment without rewriting the whole image and creating a new "wpa_supplicant.conf" i.e. resetting everything?

Thx for your help,
Solderfinger

DaciaLogan, 2019-09-09:
There are two separate  issues here:

The first time you run the program, the wpa.supplicant file makes a connection, but is then transferred to the ETC folder. (directory)  You can see this in Linux eg. on a raspberry pi 3, but you cannot see it in Windows 10. If MotionEye OS is closed and reopened, or rebooted,  the connection will normally work, but occasionally you have to open the wpa.supplicant file in the ETC folder,  and reclose it, and this normally re activates it. In practice once the camera is working it is best left on and connected.

On the second point, if you want to operate the camera on another WIFI system, then you will have to rewrite the wpa.supplicant file with the appropriate details, password etc.  It would be best to create a new sd card to do this with the fresh wpa.supplicant file.  It might be possible to delete the file in the ETC folder, and re- write it to the boot file, but I don't know if this would work.

Hope this helps.

DaciaLogan

NAVARRA, 2020-11-05:
this os is way to buggy....this is crazy.,...we have 11/2020 and still now solution???

starbasessd, 2020-11-05:
@NAVARRA Most users don't have an issue. 
If you are having an issue, please consider posting a new issue or request.
What do you consider to be a 'bug'?
What model Pi are you using?
Which version of the motionEyeOS? (20200606, dev20201026, 20190911?)
How did you image?