motioneye-project/motioneyeos #1851: Raspberry Pi Zero W interferes with Hue zigbee network

MWDelaney, 2019-03-09:
Okay this one is going to sound bananas but I promise it’s true: my Raspberry Pi Zero W running MotionEyeOS as a Fast Network Camera causes my Phillips Hue lights to become unreachable.

My setup is as follows:
* MotionEye running in a Docker container connected to the feed from the RasPi Zero W (fast network camera)
* Raspberry Pi Zero W connected to my 2.4ghz network and operating as a “fast network camera”
* Homebridge running [homebridge-ffmpeg](
https://github.com/KhaosT/homebridge-camera-ffmpeg) to make the camera available in HomeKit, and [homebridge-http-webhooks](
https://github.com/benzman81/homebridge-http-webhooks) to make the motion sensor available in HomeKit. 
* Philips Hue 2nd generation hub and about 25 hue bulbs and lights. 

The situation: everything works great until I plug my RasPi Zero W in to power. As soon as it boots up Hue lights start becoming “unreachable” in the Hue app and in HomeKit. After about 2 minutes all lights are unreachable. Unplugging the RasPi resolves the issue after a while, once the Hue system has a chance to try reconnecting with all of its bulbs. 

What I’ve tried: I’ve changed my 2.4 ghz WiFi channel, and my Hue zigbee channel to ranges that should not overlap. This did not resolve the issue. I’ve tried re-imaging the Raspberry Pi, which also did not help.

Why I think it’s MotionEye that’s causing this: prior to setting up MotionEye and MotionEyeOS, I was using [homebridge-camera-rpi](
https://github.com/moritzmhmk/homebridge-camera-rpi) which did not cause this problem. I switched to MotionEye for the additional motion sensitivity functionality. 

I know. It sounds bananas. MotionEye and Hue’s zigbee are totally unrelated. But this is what’s happening so I figured I should report it even if just to help some other poor soul who runs into this issue feel less alone. 

Thanks for reading!

ccrisan, 2019-03-10:
@MWDelaney it doesn't sound bananas at all. In fact with many devices using the 2.4Ghz band, the more traffic you have on that band, the bigger the packet loss, in general.

Considering two devices talking to each other, signals emitted by any other unrelated devices in the band will constitute *noise* for the two devices. Using frequency-hopping techniques makes connections resilient to noise, to some extent, but the higher the noise, the less reliable the connection.

Now given that motionEye emits a continuous stream of frames via WiFi, that's basically a huge amount of noise to any other devices (except for your AP, of course).

Now, the smaller the frame size, the less traffic generated by motionEye, so the less noise to your Zigbee devices.You can reduce the transmitted image frame size by reducing the:

 * image quality (recommended)
 * resolution
 * streaming framerate

MWDelaney, 2019-03-14:
This is great information. I really thought I was losing my mind.

So is my proposed setup (a few RasPi cameras feeding to a Docker "parent") just not a good idea, given the amount of constant traffic they'll generate on my network?

Would it be more practical and less noisy to set each RasPi camera up separately without the "parent" machine (and thus lose the framerate gains by doing motion-detection off the pi)?

ccrisan, 2019-03-14:
> So is my proposed setup (a few RasPi cameras feeding to a Docker "parent") just not a good idea, given the amount of constant traffic they'll generate on my network?

> Would it be more practical and less noisy to set each RasPi camera up separately without the "parent" machine (and thus lose the framerate gains by doing motion-detection off the pi)?

I would never recommend using WiFi cameras, unless there's simply no other option. So if you must use WiFi cameras, reduce the quality/resolution/framerate as much as possible.

Setting them up without a parent could be a solution for the WiFi noise, but it's definitely not a friendly setup: you'll going to have to access them one by one and that's going to be a PITA.

closed 2019-03-14 by ccrisan