motioneye-project/motioneyeos #1853: dev20190311 bug on pizero w

tigernero79, 2019-03-11:
the version dev20190311 contains several bugs on pi zero w, it does not recognize more camera coming from a price of the prerelease version 20190222 and reformatting the pizero w from zero with that version does not start going in continuous loop

zapletaldavid, 2019-03-11:
Same behavior on RPi 3B with RPi Camera V2, unable to add camera. Reverted back to dev20190223.

tigernero79, 2019-03-11:
@zapletaldavid on dev20190223 

A information to you works FAST CAMERA mode?

tigernero79, 2019-03-11:
on dev20190223 work fastnetwork caera mode image?

zapletaldavid, 2019-03-11:
OK, sorry. No, it doesn't, no picture after switching to fast network camera mode on dev20190223.

ccrisan, 2019-03-11:
Guys, the two bugs have been fixed as follows:

 * Fast network camera mode, by adf81dea0d34e7b80fa8db151a45ff2585b9308c
 * MMAL camera, by https://github.com/ccrisan/motioneye/commit/443ab40c3608b645c2b0ba8e7d93c7d42b13d09b

Thanks for reporting.

closed 2019-03-11 by ccrisan

tigernero79, 2019-03-11:
ok thanks, but always solved with version 20190311 or we have to wait for new image?

ccrisan, 2019-03-11:
Wait for the new nightly build to be ready.