motioneye-project/motioneyeos #1854: Can we change wifi credentials if we don't have access to UI?

deostroll, 2019-03-11:
The use case is, I have configured pi zero to run locally on my network. I want to take the whole thing to my friend's house and operate it there. I don't know his wifi credentials before hand. Can we still follow the steps in [wifi preconfiguration](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration) for the same.

tigernero79, 2019-03-12:
buy a Microusb Lan adapter, and connect it to the pizero port adjacent to the power supply port, put the Lan cable in it and in your network you will see IPs that it will have acquired from its router. having done this, access the web interface and modify the Wifi credentials.

jasaw, 2019-03-12:
@deostroll Load a wpa_supplicant.conf file in boot partition?

From the wifi preconfiguration page:
>Starting from version 20190119, Wifi configuration will be read every time the device boots up i.e. existing Wifi configuration will be overwritten if /boot/wpa_supplicant.conf exists. This means there is no need to reflash the SD card if you did not get the /boot/wpa_supplicant.conf correct the first time.

deostroll, 2019-03-13:
Thanks. I missed that para somehow.

closed 2019-03-13 by deostroll