motioneye-project/motioneyeos #1856: Problem with Action Button

ros9019, 2019-03-12:
Hi everyone,

i'm Daniele Rosicarelli. I've installed on my Raspberry Pi 3 Model B+ MotionEye OS the version 2019-01-19 at this link (https://github.com/ccrisan/motioneyeos/releases/download/20190119/motioneyeos-raspberrypi3-20190119.img.xz). It's very very nice OS, really beautiful.

The problem that i'm facing with is adding the action buttons at camera n°1 to turn on and off a relay which is connected a light. 
The codes that i used for turn on and off the relay are available here:

https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons

I followed exactly the instruction there reported. The problem is i cannot turn off the relay when i turn on. So only the action button for turning on the relay is working. Can you help me? I'm getting mad and i cannot understand why is not working. 

Really thanks,
best regards.

ccrisan, 2019-03-12:
1. Let us know what your action buttons files look like. We need to know their file names as well as their content.
2. Do both of them show up in the UI?
3. We need to see some logs, particularly `motioneye.log`.

ros9019, 2019-03-12:
Ho Cris,

I resolved the question releasing the pin. Now it works.
Thanks again and very compliments for this great OS :D

-Daniele Rosicarelli-

closed 2019-03-12 by ccrisan