motioneye-project/motioneyeos #1857: Slow recording fps

nopuk, 2019-03-12:
Hi,
Running a raspberry pi3 b+ on  ,OS Version | motionEyeOS 20180627, yes I know its not up to date.

Before I update to the latest, I was thinking of switching to odroid xu4(q), does this give a nice hi res, smooth recording with motion detection or is it the same as the Pi - I cannot find any review of how well it preforms with motion detection online.

0emanresuym, 2019-04-01:
I have Odroid HC1 which shares the same SoC as the XU4 and running it as a HUB, it outperforms any Raspberry.

jasaw, 2019-10-30:
Just FYI, latest nightly build now supports hardware accelerated video encoder on Odroid XU4: https://github.com/ccrisan/motioneyeos/issues/1921