motioneye-project/motioneyeos #1858: Dev20190312 same unrecognized camera error

tigernero79, 2019-03-13:
@ccrisan unfortunately also with this version dev20190312 the picamera is not recognized and if I add the camera I get an error in the attached photo. do you need any other commits?

ccrisan, 2019-03-13:
@tigernero79 I just ran a fresh dev20190312 on my RPi3 with a RPi camera attached and it showed up right away. Can you make sure that:

* you're running dev20190312 indeed (run `cat /etc/version` via SSH/putty)
* you have installed this version from scratch and not via firmware update
* you have not disconnected the camera during these tests

If the problem still persists, please attach your `motioneye.log` file.

tigernero79, 2019-03-13:
I did update from 20190119, as you can see from the attachment I run release dev20190312 but as an update. as you say I try to reinstall ds 0 and zero if it gives me the same problem.

[image]

tigernero79, 2019-03-13:
@ccrisan 

makes me understand that update does not completely overwrite all the files?

can it be solved by having other pizero far from being updated?

ccrisan, 2019-03-13:
> Does this make me realize that the update does not actually completely overwrite all the files?

It only updates the system files, not files created by the user, such as configuration files.

tigernero79, 2019-03-13:
@ccrisan 
but commits made are not system files?

if I have more zero's to be updated, how do you see that update always generates the same problem?

Is it possible that I have to reinstall everything from scratch to get the dev20190312 working?

tigernero79, 2019-03-13:
@ccrisan 
another thing that I noticed is that, it is not possible to downgrade from current versions to Latest release versions and I believe also to previous dev versions, also via USB. if I run the command from ssh example: fwupdate upgrade 20190119 while I'm in release dev20190312 the procedure is completed, but when I restart I will always have release dev20190312 and not the delta release. why?

tigernero79, 2019-03-14:
@ccrisan 
even after making a clean flash with dev20190312 the picamera at the first start is seen both in fast camera mode and in motioneye mode, as soon as I set the parameters of admin and streaming ports and other values ​​on restart the same camera image problem with prohibition, vi it is however some problem also update fails, and the downgrade is not possible. with your commits the camera is detected with a clean installation but no image is output after parameter changes, while from upload the previous dev version the camera does not show its own view.
[image]

ccrisan, 2019-03-14:
Can you attach your current config files, removing sensible information?

tigernero79, 2019-03-14:
@ccrisan 

./                                                                                                  0000755 0000000 0000000 00000000000 13432764212 007715  5                                                                                                    ustar   root                            root                                                                                                                                                                                                                   ./motioneye.conf                                                                                    0000644 0000000 0000000 00000001055 13432764211 012574  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   conf_path /data/etc
run_path /tmp
log_path /var/log
media_path /data/output
motion_binary /usr/bin/motion
log_level info
listen 0.0.0.0
port 10099
mount_check_interval 300
motion_check_interval 10
cleanup_interval 3600
remote_request_timeout 10
mjpg_client_timeout 10
mjpg_client_idle_timeout 10
smb_shares true
smb_mount_root /data/media
wpa_supplicant_conf /data/etc/wpa_supplicant.conf
local_time_file /data/etc/localtime
enable_reboot true
enable_update true
smtp_timeout 60
zip_timeout 500
add-remove-cameras true
password_hook /usr/libexec/meyepasswd
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   ./hostname                                                                                          0000666 0000000 0000000 00000000011 13432764211 011451  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   Pi_Zero_W                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       ./motion.conf                                                                                       0000666 0000000 0000000 00000000751 13432764212 012100  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   # @normal_password xxxxxxx
# @enabled on
# @show_advanced on
# @admin_username admin
# @admin_password 2xxxxxx
# @normal_username Tiger

setup_mode off
webcontrol_parms 2
webcontrol_localhost on
#thread thread-1.conf
webcontrol_html_output on
webcontrol_port 7999

#thread thread-1.conf

#thread thread-1.conf

#thread thread-1.conf

#thread thread-1.conf

#thread thread-1.conf

#thread thread-1.conf

#thread thread-1.conf

thread thread-1.conf

                       ./localtime                                                                                         0000777 0000000 0000000 00000000000 13432764211 017512  2/usr/share/zoneinfo/Europe/Rome                                                                     ustar   root                            root                                                                                                                                                                                                                   ./ssh_host_dsa_key                                                                                  0000600 0000000 0000000 00000002545 13376771750 013203  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   -----BEGIN OPENSSH PRIVATE KEY-----
b3BlbnNzaC1rZXktdjEAAAAABG5vbmUAAAAEbm9uZQAAAAAAAAABAAABsgAAAAdzc2gtZH
NzAAAAgQC1P062c/Y1A7MNvSteYpJ3nAjwkEyaM46LrEp87DCFkipyk2jjfPSsPacwUjLl
bH/xxxxxxx4Vxxx
58QAAAAScm9vdEBtZXllLThmNjVhNmVj
-----END OPENSSH PRIVATE KEY-----
[... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                     

ccrisan, 2019-03-14:
@tigernero79 please attach the separate config files (motion.conf and thread-*.conf). What you have attached in config.txt is a mess, which also happens to include an SSH private key!

tigernero79, 2019-03-14:
@ccrisan sorry
here are the 2 files required, do you need anything else?

[thread-1.txt](https://github.com/ccrisan/motioneyeos/files/2968412/thread-1.txt)

[motion.txt](https://github.com/ccrisan/motioneyeos/files/2968415/motion.txt)

tigernero79, 2019-03-14:
@ccrisan 

I would like to point out that with the stable version 20190119 everything works, even the fastboot camera mode, even with import. the
In the previous post I have attached the files you requested.

of different in those files I manually entered the string

% d-% m-% Y% H-% M-% S with the n to make the date go up other touched only access ports

[image]

ccrisan, 2019-03-14:
@tigernero79 is this from a clean dev20190312? Or did you restore configuration from previous versions?

tigernero79, 2019-03-14:
@ccrisan 
then ccrisan I did 2 tests,

the first with dev20190312 cleaned and restored from file I could not see more camera.

then redone dev20190312 cleaned and recreated all the parameters by hand after doing it and after restarting the same thing.

so I realized that something in the parameters is annoying.

another test done was to go from dev20190223 to dev20190312 where at least the image motioneyos worked but with upgrade I already wrote you it was wrong.

from clean image goes without putting options then.

ccrisan, 2019-03-15:
@tigernero79 this is becoming very difficult for me to follow. Please do only tests that are *from scratch*, with no firmware upgrades or config restores. Old configurations use `thread-*.conf` naming, while new ones use `camera-*.conf`. You should not expect a backup/restore operation to actually work.

> then redone dev20190312 cleaned and recreated all the parameters by hand after doing it and after restarting the same thing.

Let's see what is the parameter that you tweak and breaks the system. We have two options:

1. either you change each parameter one by one and see at what point stuff goes wrong,
2. or you redo your setup from scratch and then you attach your `motion.conf` and `camera-*.conf` files here so we can try to identify the issue

tigernero79, 2019-03-15:
@Ccrisan 

ok I'll be home soon and I'll try point by point until I understand what damages the image vision. however, I will then send you the 2 files you mentioned

tigernero79, 2019-03-15:
@ccrisan 

then, modified from clean installation the dev20190312 point by point, everything worked until the Video streaming section didn't enable the Authentication Mode item by setting it in basic.

as soon as this item is active, the raspberry camera stops transmitting an image and sends a camera icon with a ban.

I hope I was helpful

it's 00.12 i hope i've made good service

tigernero79, 2019-03-16:
@ccrisan read up

ccrisan, 2019-03-16:
@tigernero79 I managed to reproduce your problem and it was indeed an issue with the way motionEye's MJPEG client handles authentication to Motion. The basic authentication should now work.

However, the digest authentication still does not work with Motion 4.2 and I have no idea why. If anyone can jump in and see what's going on, I would highly appreciate it.

tigernero79, 2019-03-16:
if you release an update with. already committing for Basic authentication I would say already excellent, there is at least one protection for streaming if enabled remotely. for authentication with the Digest method I am not able to solve it personally

Deltafox79, 2019-03-16:
> @tigernero79 I managed to reproduce your problem and it was indeed an issue with the way motionEye's MJPEG client handles authentication to Motion. The basic authentication should now work. [...]

What change did you make to correct the basic authentication problem and in which file? Regards

ccrisan, 2019-03-17:
@tigernero79 

> if you release an update with. already committing for Basic authentication I would say already excellent, there is at least one protection for streaming if enabled remotely. for authentication with the Digest method I am not able to solve it personally

That's the plan. Check out the next nightly release and see if it solves your problem. In any case, thanks for your help in identifying this problem.

ccrisan, 2019-03-17:
@Deltafox79 

> What change did you make to correct the problem on basic authentication and in which file? Regards

This is the commit that fixed the basic authentication: https://github.com/ccrisan/motioneye/commit/4639502b653dfdc359087fd59e71ef11ee39c977.

And this is the function responsible for the Digest authentication: https://github.com/ccrisan/motioneye/blob/dev/motioneye/utils.py#L713.

Mr-DaveDev, 2019-03-17:
For digest, I think it needs to be HTTP 1.1  rather than HTTP 1.0

This [patch](https://gist.github.com/Mr-DaveDev/e4943cae3f091e35909e17dead18aa2d) seems to work but obviously it has some hard coded header values that need to be cleaned up.  (I'm not sure if you'll actually even need all of them)

ccrisan, 2019-03-18:
@Mr-DaveDev you're right: using HTTP/1.1 fixed the problem. Thanks for the tip.

This issue has been fixed with https://github.com/ccrisan/motioneye/commit/5b831a01b27d16fbb3556ed03f7923cf88115b91.

closed 2019-03-18 by ccrisan

tigernero79, 2019-03-18:
if you do update from 20190119 to dev20190318 cmq it does not go image. how will you solve with stable version? will still be done clean from scratch?

ccrisan, 2019-03-18:
@tigernero79 https://github.com/ccrisan/motioneyeos/blob/dev/board/common/overlay/usr/share/post-upgrade/20190307.sh should take care of the migration of config files. If it doesn't, we'll need to adjust it and make sure it actually does it well, before releasing a stable version.

tigernero79, 2019-03-19:
@ccrisan 
according to me the problem still exists, from clean installation on a pizero w with image dev20190318 and authentificazione digest enabled the picamera however it is not seen and the usual symbol of prohibition is seen.
[image]

while with basic authentication everything works

[image]

ccrisan, 2019-03-19:
@tigernero79 the Digest fix didn't make it to dev20190318. Wait for the next nightly and you should have it.

tigernero79, 2019-03-19:
will I be able to update from dev20190318 or always clean installation?

Thanks

ccrisan, 2019-03-19:
@tigernero79 a simple firmware update should be enough.

tigernero79, 2019-03-19:
@ccrisan 

update did not go well, after update camera image does not appear. and nothing served to change authentication method from Basic or digest. I think that update doesn't do its job well. dev20190318 to dev20190319

ccrisan, 2019-03-19:
@tigernero79 I will try to reproduce these upgrade problems myself.

tigernero79, 2019-03-20:
@ccrisan with last dev2190319 with installation clean from problems in the visualization of the picamera. half is seen and half is not

[image]

ccrisan, 2019-03-20:
It seems that streaming image corruptions are due to missing `Connection: close` headers. @Mr-DaveDev why do we need `Connection: close`? Isn't it more efficient to keep the connection open?

ccrisan, 2019-03-20:
@tigernero79 the nightly build coming out tomorrow will have a fix for this issue, although I'm still waiting for a reply from @Mr-DaveDev as to why we need this HTTP/1.1 + Connection: close setup.

tigernero79, 2019-03-20:
@ccrisan 

thanks for reporting, professional as always. the problem of updates between one dev and the other remains, which always makes the image unplayable.

and one is always forced to make clean installations. Great, however, your work

Mr-DaveDev, 2019-03-21:
I do not have a specific reason on why it is required.  

Motioneye was the first time that it was observed.  When I wiresharked the Firefox<->Motion for this testing I didn't see it there so I'm not sure if it is something with the Motioneye method, python, Motion implementation or something else that is requiring it.  (I only tried it after seeing it was a spec difference between HTTP 1.0 and HTTP 1.1 and when added it resolved the corrupt images.) 

When Motion gets back in its development cycle in 4Q, we could try out different options with MHD (the lib Motion uses now to process the streams and webcontrol.) to see if it can be rearranged to allow for `Connection: Keep-Alive` rather than `Connection: Close`

tigernero79, 2019-03-21:
@ccrisan 
in fastboot network camera authentication is only basic, and when it is enabled it does not come out image, I think you have to correct also ccrisan I speak of dev20190321

ccrisan, 2019-03-26:
@tigernero79 unfortunately the FNC in-browser streaming when using authentication doesn't work well in Chrome. It's not a regression introduced by the recent changes. If you inspect the HTML elements, you'll see that the MJPEG URL is supplied correctly with username and password. You'll also see that accessing the URL directly will show the stream, even in Chrome.

Firefox behaves well when it comes to MJPEG streams protected by authentication.

tigernero79, 2019-03-26:
@ccrisan ok thanks