motioneye-project/motioneyeos #1859: Adding second camema causes breakup

cjh7583, 2019-03-13:
I initially have one night vision camera attached to RP3, perfect picture day and night on 1280x720 and records fine.
As soon as I add a second camera which is just a cheap Trust brand set at 1280x720, my night vision camera starts to break up on the picture as soon as it starts to record when motion detected. 
The cheaper camera is fine with no breakup. I really want both cameras on 1280x720 so wouldn't really want to downgrade the resolution but not sure if this is the issue anyway. 
Any advice appreciated.

ccrisan, 2019-03-13:
@cjh7583 please do not ignore the new issue template and fill in the required details.

cjh7583, 2019-03-26:
Apologies I didn't see any prompts for further info when I created post. I will look again, thanks.

cjh7583, 2019-03-27:
Sorry, I have looked again...what other info do I need to fill out when creating new issue? Thanks.

cjh7583, 2019-03-31:
Is anyone able to help with this? Ive tried 2 different powered usb hubs and still I get breakup on my camera.

cjh7583, 2019-04-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 

I am running motionEyeOS version: (dev20190408).

### Board Model

I am using the following board/model: (PI 3B+).

### Camera

I am using the following type of camera: (V4L2).

My camera model is: (ELP usb 1080 camera).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3089726/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3089727/motion.log)

 * (attach e.g. boot.log)

cjh7583, 2019-04-17:
Having tried 2 different mains powered usb hubs, the issue still occurs. I think the camera is throttling the cpu but didnt think a usb camera would take that much resource. If anyone else has experienced similar then please let me know.

cjh7583, 2019-05-01:
Just want to give this a bump, thanks

closed 2020-07-24 by cjh7583